mp3guy/ElasticFusion #4: Works with XTION Pro?

mankoff, 2015-12-02:
I'm trying to use this software but may not have the right sensors...

When I run `./ElasticFusion` on a clean Ubuntu 15.04 install + `build.sh` it reports:

    Could not open "1d27/060001/6": Failed to set up USB interface!

I have an XTION Pro (not Live - No RGB). Is this the problem? I also have Kinect devices, but would love to be able to use power-over-USB. The FAQ says it can be run "only using either ICP or RGB tracking and not both", but I don't see how to enable that option. Even if I did, is the USB ID for the XTION Pro Live hard-coded? Do you know where I can purchase a Live version? There are none for sale on eBay or Amazon or other sellers.

mp3guy, 2015-12-03:
It does not work with Xtion Pro I'm afraid, it was never built to not have an RGB stream. It would need some minor code changes for that. Also you might have permissions issues on your USB, the fix for that is listed in my Logger2 repo. 

The ICP/RGB part is controlled with the ICP weight parameters. Set it to 0 for all RGB, 100 for all ICP. No idea where to get the pro live these days!

amiltonwong, 2015-12-03:
Does it work with MS Kinect?

mankoff, 2015-12-03:
No. ElasticFusion is designed for ASUS XTION Live, not Kinect. When I run it with Kinect plugged in I see:

    DeviceOpen using default: no devices found

mp3guy, 2015-12-03:
This person has branched and added freenect2 support, which is what you're after (adds Kinect support into OpenNI2) https://github.com/bckim92/ElasticFusion/commit/af77f30d675c35c1f3278ce6bdda0c3f02328d17

mankoff, 2015-12-03:
Yes just discovered that branch. Thank you! Will also look into modifying your code to work with XTION Pro. My field site doesn't have RGB anyway, so I would be turning that off when I collect data.

mp3guy, 2015-12-03:
You can also just output a log with a completely blank RGB image and make sure to set ICP weight to 100

mankoff, 2015-12-03:
Sounds good, but how do I generate that log? I'd need an XTION Pro Live, or Logger1, right? I'm working on that too...

mp3guy, 2015-12-03:
You'll have to hack the OpenNI interface in either Logger1 or Logger2 so that instead of trying to acquire an RGB stream it doesn't, and then writes an all zero buffer out for each RGB frame.

mankoff, 2015-12-03:
OK. I plan to do this in ElasticFusion directly, rather than the logger. The idea being realtime feedback of the scene will lead to better data collection. Any reason why I should do this in LoggerX instead of ElasticFusion?

mp3guy, 2015-12-03:
No, hacking the OpenNI interface in ElasticFusion will work fine too!

mankoff, 2015-12-10:
Found some Pro Live's here: http://www.ebay.com/itm/262183648648

mankoff, 2015-12-14:
So I just got my XTION Pro Live and the error persists. 

    Could not open "1d27/0600@1/4": Failed to set USB interface!

I wonder if this is a USB3 driver issue w/ Linux? Still investigating...

mp3guy, 2015-12-15:
Have you tried the permission instructions fix here? https://github.com/mp3guy/Logger2

mankoff, 2015-12-15:
That didn't fix it, but it might be related to that. I ordered 3 XTION Pro Live from the eBay link above. One shows (from `lsusb`):

    Bus 003 Device 006: ID 1d27:0600 ASUS

And the other two

        Bus 003 Device 006: ID 1d27:0601 ASUS

The `0601` product ID works. The 0600 does not. They appear identical from the outside. Anyway, it works! Thanks for the interesting software. It is a shame the sensors are so picky...

wine3603, 2017-03-17:
late reply//...... for pro live, 0600 is the older version which has some problems and 0601 was a "newer" version. 
However, both of them have problems on USB 3.0 port, ASUS released the patch to fix this on windows,
but no lucky for linux. 
For me, I found the new released driver from libfreenect2 built with openni2 can support both ASUS xtion pro live (0601) and Kinect V2. You can test after installing freenect2 by 
./NiViewer2 
In my case, both of these two sensors work fine in ElasticFusion.

closed 2022-05-27 by mp3guy