mp3guy/ElasticFusion #5: eigen/core Error

Minotaur-CN, 2015-12-03:
Hi,guys

I downloaded the ElasticFuision.git, and use build.sh to easy install ElasticFusion. The following error happened in my computer.

[  5%] Building CXX object CMakeFiles/efusion.dir/Ferns.cpp.o
In file included from /home/fhlh/Project/Backup/ElasticFusion/Core/src/Ferns.cpp:19:0:
/home/fhlh/Project/Backup/ElasticFusion/Core/src/Ferns.h:23:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
make[2]: *** [CMakeFiles/efusion.dir/Ferns.cpp.o] Error 1
make[1]: *** [CMakeFiles/efusion.dir/all] Error 2

=================================================================

I use following two ways, but no workiing.
1. sudo ln -s /usr/local/eigen3/Eigen ./
2. I checked the cmake-gui, but no Eigen PATH was found. 

I am waiting for your response asap!

Regards!
Thanks a lot!

mp3guy, 2015-12-03:
Did you run the script on a fresh install of Ubuntu 15.04 or 14.04?

Minotaur-CN, 2015-12-03:
Hi mp3guy!
I run the script not on a fresh install of Ubuntu 14.04. There was some deep learning Toolbox.
I fix the error using creating symlinks to get around it. With eigen3 installed in /usr/local/include do the following: .
----------------------------------------------------

cd /usr/local/include
sudo ln -sf eigen3/Eigen Eigen
sudo ln -sf eigen3/unsupported unsupported

------------------------------------------------
Your code is really good job! 
Thanks !

Regards!

closed 2015-12-03 by mp3guy