mp3guy/ElasticFusion #13: Runs very slowly after downloading global map

nuernber, 2016-03-01:
ElasticFusion runs very slowly after I make a call to GlobalModel::downloadMap(), e.g., by clicking on the "Save" button. I noticed that the call that makes this slow is most likely

`glGetBufferSubData(GL_ARRAY_BUFFER, 0, count * Vertex::SIZE, vertices);`

Is there a way to download the global 3D point cloud and keep the program running at the same speed after this?

mp3guy, 2016-03-01:
I was pretty sure that this was fixed here; https://github.com/mp3guy/ElasticFusion/commit/5d8ea10392f6dfb2ea2afe5b45cf88ea5633c30d

Are you definitely up to date?

nuernber, 2016-03-01:
I was at a commit after 5d8ea10392f6dfb2ea2afe5b45cf88ea5633c30d. I just updated and tried again, but it still runs slowly after pushing save.

mp3guy, 2016-03-01:
That's very strange. Is it a log file or a live capture?

nuernber, 2016-03-01:
I'm using the dyson_lab.klg file. Should I try a live capture?

mp3guy, 2016-03-01:
I can reproduce the problem myself. I'll take a look at it

nuernber, 2016-03-01:
I'll let you know if I find anything also... thanks for the quick response!

nuernber, 2016-03-02:
I just noticed that the call to indexMap.predictIndices (https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/ElasticFusion.cpp#L573) goes from taking about 1ms to 170ms. Does this help you find the problem?

nuernber, 2016-03-02:
Sorry, I really meant this line: https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/ElasticFusion.cpp#L590

nuernber, 2016-03-03:
I have a solution (that might be more hack than a real solution): create a buffer (below, `download_map_vbo`) that's identical to the other `vbos`. Then in `GlobalModel::downloadMap()`, instead of calling glGetBufferSubData with `vbos[renderSource]`, first copy `vbos[renderSource]` into `download_map_vbo[0]`, then call glGetBufferSubData with `download_map_vbo[0]`... This solution seems to work so far.

     glBindBuffer(GL_COPY_READ_BUFFER, vbos[renderSource].first);
     glBindBuffer(GL_COPY_WRITE_BUFFER, download_map_vbo[0].first);
     glCopyBufferSubData(GL_COPY_READ_BUFFER, GL_COPY_WRITE_BUFFER, 0, 0, count * Vertex::SIZE);
     glGetBufferSubData(GL_COPY_WRITE_BUFFER, 0, count * Vertex::SIZE, vertices);
`

mp3guy, 2016-03-03:
This solution does work however it runs the risk of blowing GPU memory on very large maps. I would expect OpenGL to just shift out to system memory at that point and cause a latency spike, however performance should then return to normal, which is better than just staying slow forever.

closed 2016-03-03 by mp3guy

a333klm, 2020-06-08:
@mp3guy  @nuernber 
For me it is not possible to download data every frame. With the first approach ElasticFusion needs 170ms per Frame. With the second approach executing downloadMap needs 170 ms. 

What is the intended way to access the surfelmap outside the GPU in realtime?
If this is not intended: How can I access the stable part of the surfelmap in realtime?

commit cfe7a058 (2016-03-03): Saving uses an intermediate buffer to prevent mysterious slow down while running, closes #13