mp3guy/ElasticFusion #15: Works with Microsoft Kinect?

mp3guy, 2016-03-11:
You can get it to work by adding the freenect driver to OpenNI2

git clone https://github.com/OpenKinect/libfreenect.git
cd libfreenect
mkdir build
cd build
cmake .. -DBUILD_OPENNI2_DRIVER=ON
make

ln -s lib/OpenNI2-FreenectDriver/libFreenectDriver.so ../../ElasticFusion/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers/

maq0819, 2016-03-23:
I want to run the ICL-NUIM dataset。But I don't know how to use the
 -icl command。Please tell me how to use it 。Thanks

veolata, 2016-04-28:
How accuracy we can get?  if we want to scan small but complex things under kinect's limitation?

ManyuChang, 2016-08-25:
@mp3guy   I have added the freenect driver. However I meet the following problem:

Creating live capture... failed!
DeviceOpen using default: no devices found

I use the kinect Xbox360. By " dmesg ", I found the kinect is used by PC.
Thanks in advance.

akonneker, 2016-08-25:
@ManyuChang 

Make sure that the NiViewer (or NiViewer2) application works with your kinect. If not, then it indicates there's an issue with your libfreenect driver installation.

ManyuChang, 2016-08-26:
@akonneker
Thanks for you reply. 
I have tried "sudo NiViewer" under " /OpenNI2/Source/Tools ", and the kinect works.

So?

QinZiwen, 2016-09-01:
Running GUI/build/ElasticFusion show some errors:
ElasticFusion: /home/lion/ElasticFusion/GUI/src/Tools/OpenNI2Interface.cpp:119: OpenNI2Interface::OpenNI2Interface(int, int, int): Assertion `findMode(width, height, fps) && "Sorry, mode not supported!"' failed.
已放弃 (核心已转储)

help me!

wine3603, 2016-12-09:
@akonneker  
Hi man, I am using Kinect V2, I run NiViewer2, it works well, but cannot work with ElasticFusion or Kintinuous. 
Have you tried with Kinect V2 ?

spyd3rweb, 2017-01-03:
@wine3603 
Kinect v2 Works with libfreenect2's OpenNI2 driver:
https://github.com/OpenKinect/libfreenect2

libfreenect2 supports Color and Depth video mode of 512 x 424; you'll just have to change the initialization resolution for Elastic Fusion, in MainController.cpp ln 37:
`Resolution::getInstance(512, 424);//Resolution::getInstance(640, 480)`

wine3603, 2017-02-09:
@spyd3rweb thanks, my Kintinuous and ElasticFusion work with kinect V2 after doing so, however, both of them can not find loop closing. Neither local or global. 
Is there any problem about the different resolutions of Kinect V2? its depth (512 x 424), but rgb is (1920 x 1080). From my Niviewer2, we can see they are not matched.
[image]

wine3603, 2017-02-09:
@spyd3rweb sorry man, maybe I misunderstand the codes.  I am using libfreenect2 with openNI2 in ElasticFusion now. Then,  both input rgb and depth images should be matched (512 x 424) images, right ?

spyd3rweb, 2017-02-10:
@wine3603 Yes, ElasticFusion requires both input rgb and depth images to be the same resolution, and currently with Kinect V2, libfreenect2 is the only driver I've seen that provides, by default through registration, 512x424 video modes for both rgb and depth.

These are the simple code changes I made to ElasticFusion:

Replace line 37 in MainController.cpp:
`Resolution::getInstance(512, 424);//Resolution::getInstance(640, 480);`

Replace line 45 in MainController.cpp:
`Intrinsics::getInstance(364.531494, 364.531494, 256.237396, 207.297302);//Intrinsics::getInstance(528, 528, 320, 240);`

I haven't gotten around to actually calibrating my Kinect V2 yet, but those are the camera intrinsics I pulled somewhere from within libfreenect2.

Below are the command options I use with varying degrees of depth and surfel confidence thresholds:
`ElasticFusion.exe -r -f -ftf `

First thing is to make sure your colors are correct; I had to use Flip RGB (-f). Without proper rgb the Ferns won't be made properly and the frame-to-frame RGB tracking (-ftf) won't work well. I have had more/better loop closures with ftf enabled. With ftf the tracking is obviously better, but for some reason the quality of the global model with it turned on doesn't seem to be quite as high quality as with it turned off.  

I've had pretty good success with loop closures when ftf is enabled, but most of my experimenting has been using pre-recorded klg files from LightLogger as input for the ElasticFusion's RawLogReader and rewinding/looping.   With a beefy enough GPU and CPU I'm assuming these options should also work with LiveLogReader, but I'm sure it's highly dependent on the environment and how you're filming it.  Hope this helps, and best of luck experimenting.

wine3603, 2017-02-10:
@spyd3rweb Thanks a lot!  I modified line 37 and 45. the intrinsics I estimated with this software  [https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration](url), and its some number similar to yours. 
I am using a real Kinect to do online experiment, and im using ftf, so(3), but I have never gotten a loop closure.......  neither flip rgb or not.... 
Another thing is that, from the ply files I viewed from meshlab, I found that there are extremely big distortions.  The right angles turn to be different acute angles in different frames, is this a reason for bad loop detection?  Do you have distortion problems in the final ply maps?

spyd3rweb, 2017-02-11:
@wine3603 Yes, I had an issue with wall alignment before I started using a video light and ftf with higher confidence thresholds. Sounds like you could be losing tracking, which I would either attribute to too much motion while filming or bad lighting.  To control both of those I'm now using a video light attached to an [X-Grip](https://www.amazon.com/Opteka-Professional-Camcorder-Stabilizing-Handle/dp/B003PBB4ME).

The below numbers might give you an idea the difference looping the input makes not only in the number of deformations (closures), but also in point cloud density.

Settings: 22 Confidence threshold, 4 Depth Cutoff, 10 ICP Weight

1 Loop (2,932 Frames); 02 Deformations, 599,740 Vertices
5 Loops (11,932 Frames); 25 Deformations, 1,505,692 Vertices

wine3603, 2017-02-13:
@spyd3rweb Thanks a lot man! I bought X-Grip and the video lighter, the tracking performance is much better ! it is really helpful!
After re-calibrated the Kinect V2, the distortion is much smaller than before. Here I have one question, from [iai-kinect2](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration), I got rgb camera, depth errors, ir camera and pose four sets of calibration data. IR and rgb both have intrinsic and distortion matrix, @mp3guy  but in line 45 in MainController.cpp, its only fx, fy, cx, xy of IR sensor, where are the parameters of rgb camera in ElasticFusion?  

Another question is how do you output the numbers of loops frames and deformations? Do you display them in GUI?

mp3guy, 2017-02-13:
There is no support for different camera intrinsics/extrinsics in ElasticFusion, or distortion parameters. You will have to add that yourself. They are displayed in the GUI yes.

wine3603, 2017-02-14:
@spyd3rweb   Hi man, which software are you using to record rgb-d streams with Kinect V2 ?
I try to install LightLogger from [here](https://github.com/fsrajer/LightLogger), but it cannot find realsence and openni2 libs even they are installed.

spyd3rweb, 2017-02-16:
@wine3603 LightLogger is the software I was using to record the KLG files:

In CMake I'm not sure whether or not I had to manually set the GLUT, OPENNI2, ZLIB, and JPEG variable dirs.

Specifically For OPENNI2:
set OPENNI2_INCLUDE_DIR to %GITHUB_DIR%/OpenNI2/Include
set OPENNI2_LIBRARY to %GITHUB_DIR%/OpenNI2/Bin/x64-Debug/OpenNI2.lib
set OPENNI2_ROOT to %GITHUB_DIR%/OpenNI2

I'm only using OPENNI2 so I checked WITH_OPENNI2 and unchecked WITH_REALSENSE

Since you mentioned using iai-kinect2, I'm assuming you're on Linux, because the dependency for ROS is a pain to compile on Windows, but if you're on Windows, after building LightLogger you'll need to copy the freeglut, zlib, and OpenNI2 dll's and the OpenNI2 driver folder to the LightLogger bin directory.

Alice-kenan, 2019-02-28:
@ManyuChang  Hi, did you solve this problem? I used Kinect v1, when I run ./ElasticFusion ,Creating live capture... failed!
DeviceOpen using default: no devices found
Can you help me, thanks.

zyxrrr, 2019-03-25:
@shanjichao @Alice-kenan 
You can try it.
https://github.com/mp3guy/Kintinuous/issues/5#issuecomment-474249654

closed 2022-05-27 by mp3guy