mp3guy/ElasticFusion #17: How to run with ICL-NUIM dataset

maq0819, 2016-03-23:
I have run the -icl command with the ICL-NUM file。But nothing happened。How to use this command

mp3guy, 2016-03-23:
You need to convert the pngs to .klg format, listed here https://github.com/mp3guy/Logger1/blob/master/src/Logger.cpp#L251

closed 2016-03-23 by mp3guy

HTLife, 2016-12-07:
I developed a format converter to convert png sequence to .klg.
https://github.com/HTLife/png_to_klg/tree/master

Tested with both ICL-NUIM and TUM dataset.
Check it out!!

xiaozhuka, 2017-03-17:
@HTLife  Hi Jacky
I success compile png_to_klg on windows with Visual Studio 2013 Update 5
But when test the output file with ElasticFusion, crash in RawLogReader.cpp
tmp = fread(depthReadBuffer,depthSize,1,fp);
return tmp with 0
please help me with this

HTLife, 2017-03-17:
@xiaozhuka 
Here are some suggestions for you to check:

1. Did you tired to run ElasticFusion with provided dyson_lab.klg, and get complete result?
2. If yes, check parameters you used for png_to_klg.  Here is my setting:
`./pngtoklg -w (tum dataset root folder) -o ./test.klg -t -s 0.0002`
3. ElasticFusion could not run TUM dataset under default setting (cutoff depth and other related param. should be adjusted)
`./ElasticFusion -l (path to test.klg) -d 12 -c 3 -f`
If you still encounter core dump, feel free to discuss with me.

xiaozhuka, 2017-03-17:
@HTLife Thank you for your quick reply
I fix this bug just now, allocated depthReadBuffer is two samll
I failed to run dyson_lab.klg, I guess this dataset is two large for my computer, so I want to test on some other datasets.
Again, thanks

xxsong5, 2017-07-06:
void convert2klg(const vector<string>& vcolors, const vector<string>& vdepths, string klgfile)
 18 {
 19     FILE* log = fopen(klgfile.c_str(), "wb+");
 20     int32_t numFrames=vcolors.size();
 21     fwrite(&numFrames, sizeof(int32_t),1, log);
 22 
 23     double time=1000.0, step=(1.0/30);
 24     for(int i=0; i< vcolors.size(); ++i){
 25         time+=step;
 26 
 27         cv::Mat color=cv::imread(vcolors[i]);
 28         cv::Mat depth=cv::imread(vdepths[i],CV_LOAD_IMAGE_UNCHANGED);
 29 
 30         int64_t t64=(time*1000);
 31         int32_t coW=color.rows*color.step[0];
 32         int32_t deW=depth.rows*depth.step[0];
 33 
 34 
 35         fwrite(&t64, sizeof(int64_t), 1, log);
 36         fwrite(&deW, sizeof(int32_t), 1, log);
 37         fwrite(&coW, sizeof(int32_t), 1, log);
 38         fwrite(depth.data, deW, 1, log);
 39         fwrite(color.data, coW, 1, log);
 40 
 41         cout<<i<<'/'<<vdepths.size()<<endl;
 42     }
 43 
 44     fflush(log);
 45     fclose(log);
 46 }
 47

Viky397, 2021-01-13:
> @xiaozhuka
> Here are some suggestions for you to check:
> 
>     1. Did you tired to run ElasticFusion with provided dyson_lab.klg, and get complete result?
> 
>     2. If yes, check parameters you used for png_to_klg.  Here is my setting:
>        `./pngtoklg -w (tum dataset root folder) -o ./test.klg -s 0.0002`
> 
>     3. ElasticFusion could not run TUM/ICL-NUIM dataset under default setting (cutoff depth and other related param. should be adjusted)
>        `./ElasticFusion -l (path to test.klg) -d 12 -c 3 -f`
>        If you still encounter core dump, feel free to discuss with me.

Hi! I try to run my own .klg file with ElasticFusion but get the error: