mp3guy/ElasticFusion #18: sorry! The .klg format

maqiang6, 2016-03-23:
sorry！

I want to follow the .klg format as your  annotate in the Logger::writeData。
Then,I have two questions
1.  Must  the depthSize be 640*480*siezof(short) and the imageSize  be 640*480*3*siezof(uchar)？
2.the order of colorImage is rgb or bgr?

hpp340, 2016-03-23:
Hi,
Which type of sensor do you use, Primesense or Kinect ? I have problems finding the sensor.

maqiang6, 2016-03-23:
I use the KinectV1.0.  It can work on-line .You should have a libfreenect for KinectV1.0. Author created the project using OpenNI2 that doesn't supoort the Primesense.

hpp340, 2016-03-23:
Yeah, I also use Kinect v1. I use build.sh to easy install ElasticFusion, and no error happened. Then I install libfreenect according to https://github.com/mp3guy/ElasticFusion/issues/15 , The standalone OpenNI2 programs can use the sensor but the ElasticFusion executable still can not find the sensor. How do you install it successfully?

maqiang6, 2016-03-23:
Oh.I know what's your problem.
Have you read the file under libfreenect/OpenNI2-FreenectDriver/? You should copy your libfreenect-openni2.so and libfreenectDriver.so to your OpennI2 driver directory which is ElasticFusion-master/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers/ for ElasticFusion project

hpp340, 2016-03-23:
Thanks. But where is the  libfreenect-openni2.so ? I just find libfreenectDriver.so under lib/OpenNI2-FreenectDriver/.  I have already copy libfreenectDriver.so to ElasticFusion-master/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers/ .

mp3guy, 2016-03-23:
Hi, 

1. Yes, or whatever size the source PNGs are. 
2. This means the depth is not in the correct format. It should be unsigned short, unit of millimeters. 
3. Unit-less, just make it an increasing count (so frame one has timestamp 1, etc...). 

Tom

stephane-lb, 2016-04-06:
@hpp340  Yes same issue here, did you find a solution ? Thanks

mrkstt, 2016-06-06:
@maqiang6 have you successfully made .klg file from rgbd sequence?

QinZiwen, 2016-09-01:
I use kinect v2, How should I run the GUI demo?

fsrajer, 2016-10-13:
Just to inform anyone interested. I have coded my own Logger and I am creating valid .klg files. I am compressing depth like in Logger2 and I am using directly libjpeg to compress color. 

There were no problems. One thing to note, however, is that ElasticFusion is expecting BGR instead of RGB.

mp3guy, 2016-10-13:
Just start ElasticFusion with -f

martinruenz, 2016-10-13:
@fsrajer Is your logger open-source? Could you provide a link? Thanks! 
(How is it different from Logger2?)

fsrajer, 2016-10-13:
@martinruenz I haven't made it opensource yet, but I think that would not be a problem.

It is a lightweight tool for saving stream from Intel RealSense camera into .klg file. It is a CMakeProject and was tested on Windows with Visual Studio. 
Dependencies:
* freeglut (visualization)
* librealsense (camera)
* zlib - optional (depth compression)
* libjpeg - optional (color compression)

Let me know if you would find it useful so that I know if I should make it opensource

martinruenz, 2016-10-13:
@fsrajer Sounds awesome! I would definitely be interested.

fsrajer, 2016-10-14:
OK, I have put it here. Feel free to do anything with it. I Hope it will be useful.
[https://github.com/fsrajer/LoggerRealSense](url)

M1234Thomas, 2016-11-03:
Hello all,

I have a very basic question and sorry for that.
I do not understand how to run the program with the given .klg file.
Kindly help.

HTLife, 2016-11-21:
@M1234Thomas  
See section 6 of https://github.com/mp3guy/ElasticFusion

HTLife, 2016-11-21:
I'm also trying to write a converter for ICL-NUIM dataset.
The RGB part is fine, but depth data didn't works.

I read the depth file(.png) from ICL-NUIM by opencv function "cvLoadImage"
```
IplImage *imgDepth = 
            cvLoadImage(strAbsPathDepth.c_str(), 
                        CV_LOAD_IMAGE_UNCHANGED);
```
Each depth pixel occupy 2 bytes in IplImage array, it matches @mp3guy's comment about the depth pixel should be "unsigned short".

Therefore, I write the depth data to file directly without compression.
fwrite(imgDepth->imageData, imgDepth->imageSize, 1, logFile);

Anything wrong with my format? 
My code "https://github.com/HTLife/png_to_klg/blob/feature/main.cpp".
convertion code is start at line 35.

p.s. 
1.  I also tried to compress IplImage buffer then write to file, but I got the same result, not works.

martinruenz, 2016-12-02:
@HTLife try something like this, it is working for me:
```
cv::Mat depth = imread(pathDepth, cv::IMREAD_UNCHANGED);
depth.convertTo(depth, CV_16UC1, 1000 * depthScale);
int32_t depthSize = depth.total() * depth.elemSize();
outStream.write((char*)&depthSize, sizeof(depthSize));
```
Where `depthScale` converts you depth input to [m].

HTLife, 2016-12-06:
@martinruenz It works!!  Thanks for your reply.

HTLife, 2016-12-07:
I developed a format converter to convert png sequence to .klg.
https://github.com/HTLife/png_to_klg/tree/master

Tested with both ICL-NUIM and TUM dataset.
Check it out!!

wine3603, 2017-02-09:
@HTLife great ! can it handle the case that rgb and depth have different resolutions? (such as Kinect V2)

HTLife, 2017-02-09:
@wine3603 The "png_to_klg" is not directly related to resolution issue of ElasticFusion.
"png_to_klg" simply read images and convert it to .klg format.

Possible solution could be done by resize images first.  
Converting both RGB and Depth image to 640*480 resolution, which is ElasticFusion default resolution taken by ASUS Xtion pro live.
Then feed resized image into "png_to_klg".

zhangxiaoxuan1, 2017-10-24:
I've developed a program with OpenCV to capture depth/rgb images with RealSense sensors and generate the klg file for ElasticFusion/Kintinuous here: https://github.com/zhangxiaoxuan1/intel_image_grabber

closed 2022-05-27 by mp3guy