mp3guy/ElasticFusion #19: compileError

LiliMeng, 2016-03-25:
Hi,
     Thanks for the open source ElasticFusion. Currently it has the following installation errors. Thanks

/home/lci/workspace/ElasticFusion/Core/src/GlobalModel.cpp:265:43: error: no matching function for call to ‘Uniform::Uniform(const char [4], pangolin::OpenGlMatrix&)’
     program->setUniform(Uniform("MVP", mvp));
                                           ^

mp3guy, 2016-03-25:
I would guess you haven't built Pangolin with Eigen support.

LiliMeng, 2016-03-25:
Thanks a lot! It works and was compiled successfully!

closed 2016-03-25 by mp3guy

Godpeterliu, 2017-12-07:
@LiliMeng  I faced the same bug.  I try to build Pangolin with Eigen support by modifying the CMakelists.txt in the root dir of Pangolin.  I added the sentence "option(BUILD_PANGOLIN_EIGEN "Build support for Eigen matrix types" ON)" .  But, the change do not fix the bug.   How do you fix the bug?

Godpeterliu, 2017-12-14:
@mp3guy  I faced the same bug. I try to build Pangolin with Eigen support by modifying the CMakelists.txt in the root dir of Pangolin. I added the sentence "option(BUILD_PANGOLIN_EIGEN "Build support for Eigen matrix types" ON)" . But, the change do not fix the bug. How do you fix the bug?

94zhaofang, 2019-01-10:
@mp3guy @LiliMeng How to built Pangolin with Eigen support ? Could you help me ? Thanks a lot !

battlecry231, 2019-05-27:
@94zhaofang Maybe you need to create a symbolic link so that the library could be found:

`sudo ln -s /usr/lib/eigen3/Eigen /usr/lib/Eigen`