mp3guy/ElasticFusion #20: add icp

LiliMeng, 2016-03-25:
Hi,

           Thanks for your code, it works pretty good! But when execution, such as $./ElasticFusion -l file.klg, it has the following displays. How to add the ICP step? Thanks a lot!

Your GPU "Quadro K1000M" isn't in the ICP Step performance database, please add it
Your GPU "Quadro K1000M" isn't in the RGB Step performance database, please add it
Your GPU "Quadro K1000M" isn't in the RGB Res performance database, please add it
Your GPU "Quadro K1000M" isn't in the SO3 Step performance database, please add it

mp3guy, 2016-03-25:
Run the included GPUTest program and it will output the code you can paste into this class https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/Utils/GPUConfig.h

closed 2016-03-25 by mp3guy

LiliMeng, 2016-03-26:
Thanks a lot! It works!

zry656565, 2016-04-19:
@mp3guy 
Should I create a PR to add my code generated by GPUTest for GeForce 860M?

mp3guy, 2016-04-19:
Not necessary, but if you want I'll merge it.