mp3guy/ElasticFusion #21: Several layers on the map

LiliMeng, 2016-03-26:
Hi,

         Thanks a lot for your open-source code and helpful answers!
         It seems that the map built by ElasticFusion has several layers, and cannot overlap very well.
[image]
        The parameters are from this screenshot
[image]

maqiang6, 2016-03-28:
Yeah,I also face this problem.Several layers may caused by error accumulation. In author's article,although he use Photometric consistency to optimize the camera pose,the pose isn't very accuracy. But in his article,he use local loop closure to avoid error accumulation, so  In your scanning,maybe you can make serval local loop closure to let your result better.

mp3guy, 2016-03-28:
Looking at your screenshot, I would guess this dataset has most of the important data farther than 3m away. There's a reason 3m is the default cutoff in ElasticFusion; because the data beyond that has stark errors and is in general unreliable. 

Long forward facing "everything is quite far away" datasets are the pathological case for ElasticFusion, where something like Kintinuous will excel and I recommend. 

Nevertheless, at the price of accuracy, you can coerce the system into loop closing here by relaxing some thresholds. The current thresholds are tuned for quality and comprehensive "everything is closer than 3m" (in the case of a primesense-based depth sensor) scanning. 

Looks like you're drifting too far for a local loop closure, so first you need to convince fern keyframe matches to fire. Again this is where Kintinuous would do better since it's using DBoW. Try setting the rgbd.lastICPError to < 0.0006 and the rgbd.lastICPCount to > 1800 here https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/Ferns.cpp#L279. 

Next, you will probably have to relax the restrictions on the deformation itself, so try setting meanConsError to < 0.003 and error < 2.0 here https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/Deformation.cpp#L163. 

Hopefully this will relax things enough to allow a loop closure, but quite bluntly ElasticFusion was never designed to work on data which is plainly so bad (i.e., primesense data > 3m away). Kintinuous does better here since the loop closing is up to a keyframe-based system with SURF descriptors and the TSDF fusion frontend does a better job of smoothing out noise. You could always experiment with adding DBoW to ElasticFusion.

closed 2016-03-28 by mp3guy

LiliMeng, 2016-03-28:
Thanks a lot for your helpful and detailed suggestions! :)