mp3guy/ElasticFusion #24: Problem with Eigen

jakiboi, 2016-04-19:
Hi,
Im having a problem when compiling. I’ve tried with Eigen versions 3 and 328.
Here is my error message. Would be grateful for any help.
Thanks!

jack@jack-MacBookPro-Invalid-entry-length-DMI-table-is-broken-Stop ~/elastic/ElasticFusion-master/Core/build $ make
[  5%] Building CXX object CMakeFiles/efusion.dir/IndexMap.cpp.o
/home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp: In member function ‘void IndexMap::predictIndices(const Matrix4f&, const int&, const std::pair<unsigned int, unsigned int>&, float, int)’:
/home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp:164:42: error: invalid use of incomplete type ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
     Eigen::Matrix4f t_inv = pose.inverse();
                                          ^
In file included from /usr/include/Eigen/Core:252:0,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/Shaders/Uniform.h:23,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/Shaders/Shaders.h:25,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.h:22,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp:19:
/usr/include/Eigen/src/Core/util/ForwardDeclarations.h:219:38: error: declaration of ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
 template<typename MatrixType> struct inverse_impl;
                                      ^
/home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp: In member function ‘void IndexMap::combinedPredict(const Matrix4f&, const std::pair<unsigned int, unsigned int>&, float, float, int, int, int, IndexMap::Prediction)’:
/home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp:286:42: error: invalid use of incomplete type ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
     Eigen::Matrix4f t_inv = pose.inverse();
                                          ^
In file included from /usr/include/Eigen/Core:252:0,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/Shaders/Uniform.h:23,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/Shaders/Shaders.h:25,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.h:22,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp:19:
/usr/include/Eigen/src/Core/util/ForwardDeclarations.h:219:38: error: declaration of ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
 template<typename MatrixType> struct inverse_impl;
                                      ^
/home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp: In member function ‘void IndexMap::synthesizeDepth(const Matrix4f&, const std::pair<unsigned int, unsigned int>&, float, float, int, int, int)’:
/home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp:367:42: error: invalid use of incomplete type ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
     Eigen::Matrix4f t_inv = pose.inverse();
                                          ^
In file included from /usr/include/Eigen/Core:252:0,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/Shaders/Uniform.h:23,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/Shaders/Shaders.h:25,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.h:22,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp:19:
/usr/include/Eigen/src/Core/util/ForwardDeclarations.h:219:38: error: declaration of ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
 template<typename MatrixType> struct inverse_impl;
                                      ^
/home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp: In member function ‘void IndexMap::synthesizeInfo(const Matrix4f&, const std::pair<unsigned int, unsigned int>&, float, float)’:
/home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp:434:42: error: invalid use of incomplete type ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
     Eigen::Matrix4f t_inv = pose.inverse();
                                          ^
In file included from /usr/include/Eigen/Core:252:0,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/Shaders/Uniform.h:23,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/Shaders/Shaders.h:25,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.h:22,
                 from /home/jack/elastic/ElasticFusion-master/Core/src/IndexMap.cpp:19:
/usr/include/Eigen/src/Core/util/ForwardDeclarations.h:219:38: error: declaration of ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
 template<typename MatrixType> struct inverse_impl;
                                      ^
make[2]: *** [CMakeFiles/efusion.dir/IndexMap.cpp.o] Error 1
make[1]: *** [CMakeFiles/efusion.dir/all] Error 2
make: *** [all] Error 2

mp3guy, 2016-04-19:
That's strange, try including <Eigen/Dense> at the top of the file, or adding .eval() to the end of the .inverse() calls. Other than that I'm not sure, what version of ubuntu?

mp3guy, 2016-04-19:
Oh, also please ensure you built Pangolin with Eigen

jakiboi, 2016-04-22:
Ah! thanks so much for the help.
Building Pangolin with Eigen resolved the issue.
Im managed to get everything built on Mint - Cinnamon.
Just to realise that my i didn't have a nvidia after all :(
I think i may try using logger 1 to capture, then run the data through ElasticFusion on a faster machine.

closed 2016-04-22 by mp3guy