mp3guy/ElasticFusion #27: Issue with the Eigen dependency

aicioara, 2016-05-11:
Hello. I was wondering what version of the Eigen Library (>= x) is required for building the library. 

I see in the install command that `libeigen3-dev` is stated. However, this extra step was required 

```bash
ln -sf /usr/include/eigen3/Eigen /usr/include/Eigen
```

Now, I am having some compatibility issues 

```bash
Core/src/IndexMap.cpp:164:42: error: invalid use of incomplete type ‘const struct Eigen::internal::inverse_impl<Eigen::Matrix<float, 4, 4> >’
     Eigen::Matrix4f t_inv = pose.inverse();
                                          ^
```

I was assuming it's from the library used.

mp3guy, 2016-05-11:
Have you built Pangolin with Eigen supoprt?

aicioara, 2016-05-11:
You are correct. It is worth mentioning this in FAQ, since the error is misleading.

closed 2016-05-11 by aicioara

mp3guy, 2016-05-11:
This is already mentioned in the instructions; "It is important to build Pangolin last so that it can find some of the libraries it has optional dependencies on."