mp3guy/ElasticFusion #28: It should be specified that ElasticFusion is expecting OpenNI in a nearby folder

aicioara, 2016-05-11:
```
               "${PROGRAM_FILES}}/OpenNI2/Redist"
               "${PROGRAM_FILES}/OpenNI2"
               "${CMAKE_SOURCE_DIR}/../OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../../OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../../../OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../../../../OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../../../code/OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../../../../code/OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../deps/OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../../deps/OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../../../deps/OpenNI2/Bin/x64-Release"
               "${CMAKE_SOURCE_DIR}/../../../../deps/OpenNI2/Bin/x64-Release"
```

I had it named something else. Took 2 hours to figure it out.

mp3guy, 2016-05-11:
Since Occipital's OpenNI2 fork doesn't have a make install option, if you haven't installed an openni2-dev package via apt it has to be located manually. It's impractical to search beyond a few basic directories, so in your case I would expect people to just either provide the OPENNI2_INCLUDE_DIR and OPENNI2_LIBRARY variables to cmake at first run or via ccmake or cmake-gui.

closed 2016-05-11 by mp3guy

aicioara, 2016-05-11:
I completely agree, my point was we could save someone else few hours of debugging by making clear where a manual installation should exist or how to specify the parameters to `cmake`. I believe too many assumptions are being made.

mp3guy, 2016-05-11:
Sure, added the note.