mp3guy/ElasticFusion #31: Any possibility of BSD or Apachev2 license?

ahundt, 2016-05-17:
I'm hoping to try integrating ElasticFusion into my own BSD and/or Apache v2 licensed libraries which I use in my research but the [license attached to ElasticFusion](https://github.com/mp3guy/ElasticFusion/blob/master/LICENSE.txt) seems to be incompatible with most open source licenses. 

Is there any possibility of changing the license to a more standard flavor such as BSD and Apache v2?

mp3guy, 2016-05-17:
There are no plans to change the open source license to anything else at the moment I'm afraid. If you are interested in getting a custom license, please contact researchcontracts.engineering@imperial.ac.uk

closed 2016-05-17 by mp3guy

ahundt, 2016-05-17:
To save time, I'm guessing all options they'll have at that email address will require licensing fees? Unfortunately, I don't have the resources to purchase such a license for an open source project. :-)

Thanks anyway for your kind reply

mp3guy, 2016-05-17:
As stated in the license, use of the code is permitted for non-commercial purposes only. So if you're not using the code for commercial purposes, it can be included in whatever library you want. 

However, the original license will always apply to it and its derivatives including modifications/chopping and changing of existing code. So if you make available your library which includes exact/modified/derived code from ElasticFusion in it, the original ElasticFusion license will still apply to those parts of the release (which by definition will be non-commercial) and must be included. 

You can already see this in ElasticFusion itself as it includes derivative code from PCL, which has a BSD license. The BSD license doesn't propagate to the new original ElasticFusion code (hence the entire project is not free for commercial use), but it still persists on the code which was originally in PCL. So there is a co-existence of both licenses in some files.

Free for research and not for commercial is standard for most recently open sourced SLAM systems, e.g. [LSD-SLAM](http://www.fablitec.com/products/lsd-slam/) and [ORB-SLAM](http://webdiis.unizar.es/~raulmur/orbslam/).

ahundt, 2016-05-18:
Yes what you're saying is correct and definitely makes sense for non-commercial library users that come across this. 

To make sure I'm understanding correctly I'll summarize but IANAL of course. The left side is the "upstream" license and the right side is the "downstream" or "user" library's desired license:

BSD -> other noncommercial use = OK
BSD -> other commercial use = OK

noncommercial use only -> BSD = NOT OK (my case)
commercial & noncommercial use -> BSD = Maybe / it's complicated

noncommercial use only -> noncommercial use only = OK (I could make a new library doing this)

I'm glad many of these new SLAM papers, including yours, are publishing code for noncommercial use, that's definitely an improvement over past papers which often never published code. I hope that over time this trend will continue to shift to the model used by, for example, deep learning publications which very frequently publish with surprisingly permissive licenses!