mp3guy/ElasticFusion #32: using the core without opened window

bbodin, 2016-05-17:
Hi, 

I've been able to use the ElasticFusion Core, but only after opening a window (which is not specified in the readme). 

My code looks like that : 
```
 Resolution::getInstance(...);
    Intrinsics::getInstance(...);

  pangolin::Params windowParams;
  pangolin::CreateWindowAndBind("Main", 1280, 980, windowParams);

eFusion = new ElasticFusion();

while(true) {
   eFusion->processFrame(...);
}
```

And if I remove the pangolin calls, I got a segfault.
Is there is any workaround to avoid opening a window ? 

Thanks,
Bruno

mp3guy, 2016-05-17:
So the core relies on an OpenGL context being available in the current thread before it is constructed in order to allocate textures and buffers used in GLSL processing.

Any program that uses GLSL requires this, so if you don't want to use Pangolin to create the context you'll have to use something else like GLUT or directly interface X11. You can see how Pangolin does it here; https://github.com/stevenlovegrove/Pangolin/tree/master/src/display/device

A window isn't necessarily required, just a context.

bbodin, 2016-05-23:
Thanks.
I finally get it working without window in an ugly way.
I copy pasted the pangolin code and removed the window bit.
Cheers,
Bruno

closed 2016-05-23 by bbodin