mp3guy/ElasticFusion #33: Tag a release

aicioara, 2016-05-19:
Is it possible to tag a release. Current `master` is fine. Good name would be `v1.0.0` or `v1.1.0`

aicioara, 2016-05-27:
@mp3guy Any feedback on this?

mp3guy, 2016-05-27:
I'm not interested in versioning the code given that it's research and not production.

closed 2016-05-27 by mp3guy

aicioara, 2016-05-27:
I don't really see the difference between research and production. I think it is in the best interest of the author to allow other people depend on the code.

I would like to take this project as a dependency for my project and reference back to it. The alternative is to fork it version it myself, which I would prefer not to, in appreciation for author's work.

mp3guy, 2016-05-27:
There's a pretty big difference, and it could also complicate licensing. People are free to the depend on the code if they wish (the license will always persist). Not having explicit versioning makes sense since bugfixes and updates will always sit on top of master and there are no plans for big revisions to the project itself. So if you depend on ElasticFusion, you depend on top of master. 

You can do that, I encourage forking if people want to.

aicioara, 2016-05-27:
Versioning helps with backwards compatibility. That is my only concern.

mp3guy, 2016-05-27:
If an older revision is required it can be identified by the commit hash, similar to how I do it with DBoW in Kintinuous.

aicioara, 2016-05-27:
A tag is just naming a commit hash. It has no further implications. Makes the repository look more professional and better maintained. 

As in here https://github.com/stevenlovegrove/Pangolin/releases

mp3guy, 2016-05-27:
It's not a professional repository, it's a free for non-commercial use open source release of a research project. 

Pangolin is continuously in flux and you'll notice that Steve only ever did one such release and hasn't since, most of the development happens on the devel branch and many other SLAM systems depend on Pangolin and I don't recall any of them referencing that specific v0.3 tag.