mp3guy/ElasticFusion #35: support for nvidia gpu version 330

ahundt, 2016-05-19:
What's necessary to enable a gpu with version 330 (Geforce GT 750M 2048MB)?

```
./ElasticFusion
Creating live capture... OpenNI2-FreenectDriver: Using libfreenect v0.5.2
failed!
DeviceOpen using default: no devices found
GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/draw_global_surface.vert:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version
ERROR: 0:21: 'layout' : syntax error: syntax error

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/draw_global_surface.geom:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version
ERROR: 0:21: 'layout' : syntax error: syntax error

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/draw_global_surface_phong.frag:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version
ERROR: 0:26: 'flat' : syntax error: syntax error

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/empty.vert:
ERROR: 0:19: '' :  version '330' is not supported

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/quad.geom:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version
ERROR: 0:21: 'layout' : syntax error: syntax error

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/fxaa.frag:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/empty.vert:
ERROR: 0:19: '' :  version '330' is not supported

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/quad.geom:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version
ERROR: 0:21: 'layout' : syntax error: syntax error

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/resize.frag:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/empty.vert:
ERROR: 0:19: '' :  version '330' is not supported

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/quad.geom:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version
ERROR: 0:21: 'layout' : syntax error: syntax error

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/resize.frag:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/empty.vert:
ERROR: 0:19: '' :  version '330' is not supported

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/quad.geom:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version
ERROR: 0:21: 'layout' : syntax error: syntax error

GLSL Shader compilation failed: /Users/athundt/source/git/ElasticFusion/Core/src/Shaders/resize.frag:
ERROR: 0:19: '' :  version '330' is not supported
ERROR: 0:19: '' : syntax error: #version
```

mp3guy, 2016-05-19:
Either your OS isn't setting your nVidia GPU as default, so you'll need to use nVidia prime to set it, or your OpenGL install doesn't support that version. According to nVidia that card supports up to 4.5, so it must be your OS/drivers.

ahundt, 2016-05-19:
hmm, I haven't gone deeply into GPU programming yet so forgive me if I'm mistaken about something. I'm able to use many other libraries without issue such as caffe and I have cuda 7.5 installed. What is 4.5 the version of? I'm also not familiar with nvidia prime, I'll take a look at what that is when I have a moment. Thanks

stephane-lb, 2016-05-20:
@ahundt If I understood you are on MacOS so in that case there is many issue to declare the context for OpenGL 4.3 or 3.3, I would have a look at this issue as soon as I would be able to compile on mine. I suppose that you should be able to use #version 150 in case of #version 330 it is the same except that it should not flag an error.

mp3guy, 2016-05-21:
Using 150 won't work since the shaders are actually written for a later version of OpenGL. 4.5 is the version of OpenGL (which would translate to a shader version of 450). 

But you should investigate what Stephane suggested about declaring the context for OSX, something I'm not familiar with.

stephane-lb, 2016-05-22:
@ahundt : just commit and push the missing files on your repo and I will help you to make the investigation on this issue.

ahundt, 2016-06-01:
@stephane-lb thanks I've pushed what I have to https://github.com/ahundt/ElasticFusion I've also enabled issues over there. Sorry for the delay between replies, I'm currently traveling.

stephane-lb, 2016-06-01:
@ahundt I will check this afternoon, I am going for a long trip today, so I might not have time for that before july ...

stephane-lb, 2016-09-16:
just as a reference : https://github.com/stevenlovegrove/Pangolin/issues/145

closed 2022-05-27 by mp3guy