mp3guy/ElasticFusion #37: Does it worked for windows 7 Operating System?

kerrylev, 2016-06-01:
You have did a great work, it's amazing! And I was wondering if there had been any more consideration for windows OS, all the dependencies can be build in windows OS. Do you ever have a try ?

mp3guy, 2016-06-01:
I haven't tried it myself, but I've heard that it builds ok on Windows.

fsrajer, 2016-10-07:
@kerrylev, check my pull request.

xiaozhuka, 2017-03-21:
I have successfully compiled on Windows with VS 2013, but there always some unkonwn error when running with ICL-NUIM dataset.

fsrajer, 2017-03-21:
I didn't run it with that dataset. Feel free to patch my changes.

LittleLampChen, 2018-09-26:
@fsrajer I have try to compile core code in windows, but failed with the error 
""
[image]

closed 2022-05-27 by mp3guy