mp3guy/ElasticFusion #46: running ./GUI/build/ElasticFuision show error with kinect2

QinZiwen, 2016-09-01:
My computer installed Ubuntu14.04, and  building is nice, but ./GUI/build/ElasticFuision show some message as following:
Creating live capture... [Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 021468743547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 021468743547
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
ElasticFusion: /home/lion/ElasticFusion/GUI/src/Tools/OpenNI2Interface.cpp:119: OpenNI2Interface::OpenNI2Interface(int, int, int): Assertion `findMode(width, height, fps) && "Sorry, mode not supported!"' failed.
已放弃 (核心已转储)
what should I do?
Thank you!

closed 2016-09-01 by mp3guy

spyd3rweb, 2017-01-03:
@QinZiwen 
libfreenect2 supports Color and Depth video mode of 512 x 424; you'll just have to change the initialization resolution for Elastic Fusion, in MainController.cpp ln 37:
`Resolution::getInstance(512, 424);//Resolution::getInstance(640, 480)`

ganlumomo, 2017-03-10:
Also Intrinsics::getInstance(604, 604, 515, 268); or provide a calibration file

shanjichao, 2018-04-10:
@spyd3rweb @ganlumomo hello,I have modified the MainController.cpp with Resolution::getInstance(640, 480) and Intrinsics::getInstance(604, 604, 515, 268),but cannot solve it ,please help me ,thanks

benlee12, 2018-10-26:
@ganlumomo thanks , helps a lot

Fosurix, 2021-05-11:
If you are using Kinect V2, then the intrinsics should be about (338,339,253,214)