mp3guy/ElasticFusion #49: issue on cuda 8.0

fyj3266098, 2016-09-30:
Creating live capture... Warning: USB events thread - failed to set priority. This might cause loss of data...
success!
Waiting for first frame...... got it!
OpenGL Error: XX (1282)
In: /usr/local/include/pangolin/gl/gl.hpp, line 198
Error: invalid texture reference: /home/pc/PC/ElasticFusion-master/Core/src/Cuda/convenience.cuh:68

any advice will be appreciated

mp3guy, 2016-09-30:
I'm afraid I cannot reproduce on Ubuntu 16.04, gcc 5.4, nvcc 8.0.44, Pangolin@204ab84b21e0dacb2dcfb85936f21b1d28e9fc4d master

mp3guy, 2016-09-30:
If you're running it without the GUI, you must make sure you have created an OpenGL context before creating an instance of ElasticFusion, that might be your problem.

aivision2020, 2016-10-05:
I'm also having issues ubuntu 14, cuda 8.0

(env1)agolbert@ildl088:~/ElasticFusion/GUI/build$ ./ElasticFusion -i dyson_lab.klg 
Creating live capture... failed!
DeviceOpen using default: no devices found

I purged and reinstalled cuda toolkit and nvidia drivers

Thanks

M1234Thomas, 2016-11-06:
Hello,

I am getting an error as follows.

OpenGL Error: XX (1282)
In: \ElasticFusion\Pangolin-master\include\pangolin/gl/gl.hpp, line 562

I have checked the graphics card and is set to use NVIDIA to use all the times. 
I can not figure out what the issue is.

closed 2022-05-27 by mp3guy