mp3guy/ElasticFusion #50: Modify Point Weight in images

TsinghuaPhantom, 2016-10-04:
Hello, your project is very helpful. I have a question here, is it possible to modify the code that allow some area in the image has more weight during reconstruction, says these areas require less error during bundle adjustment. If can, which file should I modify, thank you very much!

mp3guy, 2016-10-05:
There is no bundle adjustment in ElasticFusion. Do you mean more weight during reconstruction or during tracking or both?

TsinghuaPhantom, 2016-10-05:
Yes I mean both. Thank you.

mp3guy, 2016-10-05:
To change reconstruction weight you can alter how the confidence is computed here; https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/Shaders/surfels.glsl#L36

To change tracking weight it's more involved, you'll need to modify both CUDA kernels related to the tracking reductions https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/Cuda/reduce.cu#L347 and https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/Cuda/reduce.cu#L533

closed 2016-10-05 by mp3guy