mp3guy/ElasticFusion #52: Error: invalid texture reference

againxx, 2016-10-09:
Hi:
   I'm using a new laptop with GTX1080 to test this code. I have installed the cuda-7.5 by runfile. Because if I install cuda through apt-get, it will uninstall my latest nvidia driver 370.28 and install 361 which do not support GTX1080. With all this finished, I met an error:
Error: invalid texture reference: /home/ustc-1314/ElasticFusion/Core/src/Cuda/convenience.cuh:68

Any advice will be appreciated!

againxx, 2016-10-09:
I just used the command in GUI
./ElasticFusion -l dyson_lab.klg

mp3guy, 2016-10-09:
Pretty sure you need CUDA 8 for the 1080

againxx, 2016-10-10:
I just reinstalled cuda 8.0 and the result of deviceQuery is something like this:

./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

Detected 1 CUDA Capable device(s)

Device 0: "GeForce GTX 1080"
  CUDA Driver Version / Runtime Version          8.0 / 8.0
  CUDA Capability Major/Minor version number:    6.1
  Total amount of global memory:                 8105 MBytes (8498774016 bytes)
  (20) Multiprocessors, (128) CUDA Cores/MP:     2560 CUDA Cores
  GPU Max Clock rate:                            1734 MHz (1.73 GHz)
  Memory Clock rate:                             5005 Mhz
  Memory Bus Width:                              256-bit
  L2 Cache Size:                                 2097152 bytes
  Maximum Texture Dimension Size (x,y,z)         1D=(131072), 2D=(131072, 65536), 3D=(16384, 16384, 16384)
  Maximum Layered 1D Texture Size, (num) layers  1D=(32768), 2048 layers
  Maximum Layered 2D Texture Size, (num) layers  2D=(32768, 32768), 2048 layers
  Total amount of constant memory:               65536 bytes
  Total amount of shared memory per block:       49152 bytes
  Total number of registers available per block: 65536
  Warp size:                                     32
  Maximum number of threads per multiprocessor:  2048
  Maximum number of threads per block:           1024
  Max dimension size of a thread block (x,y,z): (1024, 1024, 64)
  Max dimension size of a grid size    (x,y,z): (2147483647, 65535, 65535)
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and kernel execution:          Yes with 2 copy engine(s)
  Run time limit on kernels:                     Yes
  Integrated GPU sharing Host Memory:            No
  Support host page-locked memory mapping:       Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support:                        Disabled
  Device supports Unified Addressing (UVA):      Yes
  Device PCI Domain ID / Bus ID / location ID:   0 / 1 / 0
  Compute Mode:
     < Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) >

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version = 8.0, CUDA Runtime Version = 8.0, NumDevs = 1, Device0 = GeForce GTX 1080
Result = PASS

But I still have the error:
ustc-1314@ustc1314-msi:~/ElasticFusion/GUI/src/build$ ./ElasticFusion -l dyson_lab.klg
Error: invalid texture reference: /home/ustc-1314/ElasticFusion/Core/src/Cuda/convenience.cuh:68

mp3guy, 2016-10-10:
Did you build with CUDA_NVCC_ARCHS listing 61?

againxx, 2016-10-10:
Sorry, I'm a freshman to CUDA, I just followed the instructions to install CUDA8, OpenNI2 and Pangolin. So I don't know what CUDA_NVCC_ARCHS mean. Could you please tell me more detail about this? Is this a parameter in CMakeLists or somewhere else？

mp3guy, 2016-10-10:
Yes it's a CMake parameter in the core build

againxx, 2016-10-11:
It works! Thank you very much!

closed 2016-10-11 by againxx