mp3guy/ElasticFusion #55: Problem to run demo dataset

HTLife, 2016-10-17:
The hardware that I use is same with this issue post (https://github.com/mp3guy/Kintinuous/issues/15)

I think this error message should also relate with CUDA and my GPU
```
$ ./ElasticFusion -l dyson_lab.klg
Error: invalid texture reference: /home/(username)/code/ElasticFusion/Core/src/Cuda/convenience.cuh:68
```

HTLife, 2016-10-17:
For someone also meet this problem

ElasticFusion/Core/src/CMakeLists.txt

add "61" to Line24:  
```
set(CUDA_ARCH_BIN "20 30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch     to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) o    r 13 21(13)")
```

closed 2016-10-17 by HTLife

zss1995, 2019-03-08:
> For someone also meet this problem
> 
> ElasticFusion/Core/src/CMakeLists.txt
> 
> add "61" to Line24:
> 
> ```
> set(CUDA_ARCH_BIN "20 30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch     to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) o    r 13 21(13)")
> ```

Have you solved this problem,I met the same