mp3guy/ElasticFusion #57: GTX 1080 parameters

YangLiu-DayDreammer, 2016-10-20:


HTLife, 2016-10-20:
Did you successfully run sample dataset?

Maybe your problem is related to mine (https://github.com/mp3guy/ElasticFusion/issues/55).

YangLiu-DayDreammer, 2016-10-20:
Thank you very much. It perfectly solve my problem. I think for GTX 1060/1070/1080 and other future GPUs we should modify CMakeLists to adjust.

closed 2016-10-20 by YangLiu-DayDreammer

zssjh, 2018-07-26:
@HTLife I git the newest version, and the cmakelist.txt have been modified(listing 61),but I still face the problem, what is the reason of the problem? many many thanks!

luruibo1995, 2019-04-17:
The Compute Capability of your GPU is somewhat important. My GPU is 2080 and I have to update cuda to the 10+, add 75 in the CMakeLists.txt just like  #55 and modify the  function __shfl_down to __shfl_down_sync , the mask i choice is 0xFFFFFFFF, you can refer https://docs.nvidia.com/cuda/cuda-c-programming-guide/index.html#synopsis and https://stackoverflow.com/questions/50639194/shfl-down-and-shfl-down-sync-give-different-results.