mp3guy/ElasticFusion #61: opengl error 1282

M1234Thomas, 2016-11-06:
Hello,

I am getting an error as follows.

OpenGL Error: XX (1282)
In: \ElasticFusion\Pangolin-master\include\pangolin/gl/gl.hpp, line 562

I have checked the graphics card and is set to use NVIDIA to use all the times. 
I can not figure out what the issue is.

mp3guy, 2016-11-06:
Need your OS and graphics card to have any idea of what's going on.

M1234Thomas, 2016-11-06:
I am using visual studio 2013 on windows 7 with GeForce gtx 750 ti Graphics card.

mp3guy, 2016-11-06:
I've never run it on windows before so I don't have any idea what the problem might be.

M1234Thomas, 2016-11-06:
Could you please brief me if the error was on linux, what could have been the possible cause for this kind of error?

mp3guy, 2016-11-06:
OpenGL error 1282 is invalid operation. If you look at that link it tells you; https://github.com/stevenlovegrove/Pangolin/blob/master/include/pangolin/gl/gl.hpp#L562, something weird is happening when trying to bind a target to a frame buffer.

closed 2016-11-22 by M1234Thomas

Buzzroid, 2017-07-08:
See the @fsrajer comment at https://github.com/stevenlovegrove/Pangolin/issues/149
It fixes this problem