mp3guy/ElasticFusion #63: cmd.exe exited with code 1..

antithing, 2016-11-19:
Hi, thank you for making this code available, it looks great. I am attempting to build on Windows 10, have linked all the dependencies, but when i try to build libefusion.sln, I get the error:

'cmd.exe exited with code 1' and no other info. What could I be missing here that I need to change?

Thanks again!

antithing, 2016-11-20:
Ok, using '20' in the `CUDA_ARCH_BIN` field has fixed this. Moving to vs2013 fixed the next error I saw, and reinstalling CUDA somewhere without any spaces in the folder path fixed the next one. Now, i am close!

I am getting this error:

`Error	3661	error C2665: 'Uniform::Uniform' : none of the 7 overloads could convert all the argument types	
`
In `GlobalModel.cpp`, at the line

`   program->setUniform(Uniform("MVP", mvp));`

Do I need to adjust something? or do i still have some linkage errors?

Thank you again!

mp3guy, 2016-11-20:
I don't know too much about running it on Windows, you're better off asking https://github.com/fsrajer

antithing, 2016-11-20:
Thanks! I will ask him.

antithing, 2016-11-20:
I actually just now got this building! Recompiling everything (Pangolin, suitesparse etc) in vs2013 seems to work.

Building the GUI gives me a new error though:

`Error	5363	error LNK2005: glewInit already defined in efusion.lib(efusion.dll)	`

Would you be able to point me in the right direction to fix this?

Thanks again!

yylovess, 2016-11-21:
@antithing 
sorry to bother you, but I have the same problem when compiling elastic/fusion core in VS2013
I am getting this error:

Error 3661 error C2665: 'Uniform::Uniform' : none of the 7 overloads could convert all the argument types 
3968 IntelliSense: no instance of constructor "Uniform::Uniform" matches the argument list argument types are: (const char [4], pangolin::OpenGlMatrix)
In GlobalModel.cpp, at the line

program->setUniform(Uniform("MVP", mvp));

could you please tell me you solution.
Thank you very much.

antithing, 2016-11-21:
Hi, sure!

What worked for me was recompiling all the dependencies in vs2013, and rebuilding the Core.

yylovess, 2016-11-21:
@antithing 
Thank you very much , did you build eigen from resource or just install it from .exe file.
The Pangolin should be compiled with eigen ,could I do it by add the eigen include dir ?

antithing, 2016-11-21:
Eigen is a header-only library. You do not need to build it.

Yes, add manually the entries in cmake to build pangolin with Eigen.

yylovess, 2016-11-21:
@antithing thank you very much and i will try now

syncle, 2017-01-07:
I added following code in /core/src/shaders/uniform.h

	Uniform(const std::string & id, const pangolin::OpenGlMatrix & v)
         : id(id),
           t(MAT4)
        {
	   for(int i=0; i<4; i++)
	     for(int j=0; j<4; j++)
	       m4(i,j) = v(i,j);	   
	}

xiaozhuka, 2017-03-20:
hello @antithing 
When I compile efusion lib in vs2013, there is fatal error about nvcc

1>  Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/Cuda/Debug/cuda_compile_1_generated_cudafuncs.cu.obj
1>  nvcc fatal   : Could not open output file 'C:\Users\LHY\AppData\Local\Temp;D:\Python27/tmpxft_00009044_00000000'
1>  
1>  CMake Error at cuda_compile_1_generated_cudafuncs.cu.obj.Debug.cmake:222 (message):
1>    Error generating
1>    E:/coding.net/ElasticFusionDevelop/Core/build/CMakeFiles/cuda_compile_1.dir/Cuda/Debug/cuda_compile_1_generated_cudafuncs.cu.obj

do you know how to fix this?

smmmmi, 2017-04-20:
@xiaozhuka 
Same error! How did you fix it?

closed 2022-05-27 by mp3guy