mp3guy/ElasticFusion #70: Compilation errors for glewInit already defined, and GL calls result in Access Violation

spyd3rweb, 2016-12-30:
Compilation warning/error(s):
Error	LNK2005	glewInit already defined in efusion.lib(efusion.dll)	ElasticFusion	C:\Users\spyde\Documents\GitHub\ElasticFusion\GUI\src\build\glewd.lib(glew.obj)	1
Error	LNK1169	one or more multiply defined symbols found	ElasticFusion	C:\Users\spyde\Documents\GitHub\ElasticFusion\GUI\src\build\Debug\ElasticFusion.exe	1

After suppressing this error (build details below), running ElasticFusion.exe results in:
Exception thrown at 0x00000000 in ElasticFusion.exe: 0xC0000005: Access violation executing location 0x00000000
The culprit being:
glGenRenderbuffersEXT(1, &rbid) in ElasticFusion.exe!pangolin::GlRenderBuffer::GlRenderBuffer(int width, int height, int internal_format) Line 434	C++

Various attempts to troubleshoot included using glew static library linking and shared library linking from Pangolin which have allowed gl calls from within ElasticFusion.exe to proceed further into execution; however, eventually gl calls originating from efusion.dll would then fail with an access violation.

Could you please validate build instructions and appropriate glew linkage for ElasticFusion and libefusion. 

Likely unrelated, but the line ElasticFusion.exe!pangolin::GlRenderBuffer::GlRenderBuffer(int width, int height, int internal_format) Line 434 appears not to be properly terminated. Has "," instead of ";"

If the problem persists after I have confirmation from you that linkage within ElasticFusion and libefusion is corrct for glew, I will create an issue with the maintainer of Pangolin. There could be a potential issue for:
http://www.khronos.org/opengl/wiki/OpenGL_Loading_Library#Intialization_of_GLEW_1.13.0_and_earlier
GLEW up to version 1.13.0 has a problem with core contexts. It calls glGetString*GL_EXTENSIONS) which causes GL_INVALID_ENUM on GL 3.2+ core context as soon as glewInit() is called. It also doesn't fetch the function pointers. GLEW version 2.0.0+ uses glGetStringi instead. The only fix for earlier version is to use glewExperimental. If you are creating a GL context the old way or if you are creating a backward compatible context for GL 3.2+ then you don't need glewExperimental.
glewExperimental = GL_TRUE;
GLenum err=glewInit();

Build and Run procedures:
# Suite Sparse for Windows 1.3.1
1. Extract to GitHub folder
2. Modify checkGetSuiteSparse.cmake to download latest SuiteSparse (4.5.4)
#set(SUITESPARSE_URL "http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-4.4.3.tar.gz")
set(SUITESPARSE_URL "http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-4.5.4.tar.gz")
3. Download latest verson of Metis v4 library (4.0.3) from http://glaros.dtc.umn.edu/gkhome/fsroot/sw/metis/OLD
4. Merge downloaded Metis with Metis folder under Suite Sparse for Windows (https://github.com/jlblancoc/suitesparse-metis-for-windows/issues/7)
5. Configure CMAKE, Visual Studio 14 2015 Win64 generator (http://stackoverflow.com/questions/33097558/cmake-cuda-libraries-not-found-when-compiling-opencv)
- set EXECUTABLE_OUTPUT_PATH %GITHUB_DIR%/suitesparse-metis-for-windows/bin
- set LIBRARY_OUTPUT_PATH %GITHUB_DIR%/suitesparse-metis-for-windows/lib
- set SUITESPARSE_INSTALL_PREFIX %GITHUB_DIR%/suitesparse-metis-for-windows/
- set WITH_CUDA true
- set HAVE_COMPLEX false (https://github.com/jlblancoc/suitesparse-metis-for-windows/issues/5)
6. Compile and wait for errors: 
- change long unsigned int -> size_t
- cast void* to (unsigned char *)
- add typedef int(*__compar_fn_t) (const void *, const void *);
- add C:\Users\spyde\Documents\GitHub\suitesparse-metis-for-windows\SuiteSparse\GPUQREngine\Include to list of include directores for spqr
- add C:\Users\spyde\Documents\GitHub\suitesparse-metis-for-windows\SuiteSparse\SuiteSparse_GPURuntime\Include to list of include directores for spqr
7. Build x64
8. Install

# OpenNI2 Kinect2 Branch (I understand Kinect 2 is not currently supported by ElasticFusion, but once I get ElasticFusion running, I plan on creating a Kinect2 OpenNI2 driver that provides a color and depth of the same resolution)
1. Extract to GitHub folder
2. Unload Kinect project
3. Set Kinect 2 project properties for Linker to /WX:No for treat link warning as errors. LNK4075 ignoring '/EDITANDCONTINUE' due to '/OPT:LBR' specification	
4. Change all project properties for C/C++ to compile with /W3 and /WX- to avoid warnings as errors
5. Build x64

# Pangolin
1. Extract Eigen3 to GitHub folder
2. Extract Pangoline to GitHub folder
3. Configure CMAKE
- set MSVC_USE_STATIC_CRT false
- set OPENNI2_LIBRARY %GITHUB_DIR%/OpenNI2/Bin/x64-Debug/OpenNI2.lib
- validate EIGEN3_INCLUDE_DIR
- set CMAKE_INSTALL_PREFIX %GITHUB_DIR%/Pangolin
4. Build x64
5. Install

# Elastic Fusion Core
1. Extract Elastic Fusion to GitHub folder
2. Configure CMAKE for Core
- set SUITESPARSE_INCLUDE_DIR to %GITHUB_DIR%/suitesparse-metis-for-windows/include/suitesparse
- set SUITESPARSE_LIBRARY_DIR to %GITHUB_DIR%/suitesparse-metis-for-windows/lib64
- set SUITESPARSE_SPQR_VALID true
- validate SUITESPARSE_XXXX_LIBRARY
- validate SUITESPARSE_SPQR_LIBRARY
- set CMAKE_INSTALL_PREFIX %GITHUB_DIR%/ElasticFusion
3. Build x64
4. Install

# Elastic Fusion GUI
1. Skip -- Extract Elastic Fusion to GitHub folder
2. Configure CMAKE for GUI
- set JPEG_LIBRARY %GITHUB_DIR%/Pangolin/build/external/libjpeg/lib/jpeg.lib
- set JPEG_INCLUDE_DIR %GITHUB_DIR%/Pangolin/build/external/libjpeg/include
- create BLAS_LIBRARIES_DIR %GITHUB_DIR%/suitesparse-metis-for-windows/lib64/lapack_blas_windows
- create LAPACK_LIBRARIES_DIR %GITHUB_DIR%/suitesparse-metis-for-windows/lib64/lapack_blas_windows
- set ZLIB_LIBRARY %GITHUB_DIR%/Pangolin/build/external/zlib/lib/zlibd.lib
- set ZLIB_INCLUDE_DIR %GITHUB_DIR%/Pangolin/build/external/zlib/include
- set OPENNI2_LIBRARY %GITHUB_DIR%/OpenNI2/Bin/x64-Debug/OpenNI2.lib
- set SUITESPARSE_INCLUDE_DIR %GITHUB_DIR%/suitesparse-metis-for-windows/include/suitesparse
- Set SUITESPARSE_LIBRARY_DIR to %GITHUB_DIR%/suitesparse-metis-for-windows/lib64
- validate SUITESPARSE_XXXX_LIBRARY
- set EFUSION_LIBRARY %GITHUB_DIR%/ElasticFusion/lib/efusion.lib
- set CMAKE_INSTALL_PREFIX %GITHUB_DIR%/ElasticFusion
3. Add /FORCE to linker command line option to get rid of: 
SeverityCode	Description	Project	File	Line	Suppression State	
Error	LNK2005	glewInit already defined in efusion.lib(efusion.dll)	ElasticFusion	C:\Users\spyde\Documents\GitHub\ElasticFusion\GUI\src\build\glewd.lib(glew.obj)	1
Error	LNK1169	one or more multiply defined symbols found	ElasticFusion	C:\Users\spyde\Documents\GitHub\ElasticFusion\GUI\src\build\Debug\ElasticFusion.exe	1	
4. Build x64
5. Install
6. Copy required dlls
copy "%GITHUB_DIR%\Pangolin\build\external\zlib\bin\zlibd.dll" "%~dp0"
copy "%GITHUB_DIR%\OpenNI2\Bin\x64-Debug\OpenNI2.dll" "%~dp0"
Xcopy /E /I "%GITHUB_DIR%\OpenNI2\Bin\x64-Debug\OpenNI2\OpenNI2" "%~dp0"
copy "%GITHUB_DIR%\suitesparse-metis-for-windows\lib64\lapack_blas_windows\libblas.dll" "%~dp0"
copy "%GITHUB_DIR%\suitesparse-metis-for-windows\lib64\lapack_blas_windows\liblapack.dll" "%~dp0"
copy "%GITHUB_DIR%\suitesparse-metis-for-windows\lib64\lapack_blas_windows\libgcc_s_sjlj-1.dll" "%~dp0"
copy "%GITHUB_DIR%\suitesparse-metis-for-windows\lib64\lapack_blas_windows\libgfortran-3.dll" "%~dp0"
copy "%GITHUB_DIR%\suitesparse-metis-for-windows\lib64\lapack_blas_windows\libquadmath-0.dll" "%~dp0"
7. Run ElasticFusion.exe
Exception thrown at 0x00000000 in ElasticFusion.exe: 0xC0000005: Access violation executing location 0x00000000
Call Stack:
>	ElasticFusion.exe!pangolin::GlRenderBuffer::Reinitialise(int width, int height, int internal_format) Line 446	C++
 	ElasticFusion.exe!pangolin::GlRenderBuffer::GlRenderBuffer(int width, int height, int internal_format) Line 434	C++
 	ElasticFusion.exe!GUI::GUI(bool liveCap, bool showcaseMode) Line 55	C++
 	ElasticFusion.exe!MainController::MainController(int argc, char * * argv) Line 113	C++
 	ElasticFusion.exe!main(int argc, char * * argv) Line 23	C++
 	ElasticFusion.exe!invoke_main() Line 65	C++
 	ElasticFusion.exe!__scrt_common_main_seh() Line 253	C++
 	ElasticFusion.exe!__scrt_common_main() Line 296	C++
 	ElasticFusion.exe!mainCRTStartup() Line 17	C++

spyd3rweb, 2016-12-31:
Until you hear back from @mp3guy, the interim solution for me was to modify the Shader constructor to  have pangolin check for a current context, create one if it didn't exist, and call glewInit();
```
/*
Function to get, find, or initialize Pangolin context and initialize glew
*/

#if !defined(__STATIC_GLEW_INIT_HPP__)
#define __STATIC_GLEW_INIT_HPP__
#define GLEW_STATIC
#include <pangolin/pangolin.h>
#include <pangolin/gl/gl.h>
#include <pangolin/gl/glplatform.h>
#include <pangolin/display/display_internal.h>
#include <string>

static inline void staticGlewInit(std::string name = "Main")
{
	// GetCurrentContext
	pangolin::PangolinGl* context = pangolin::GetCurrentContext();
	if (context == NULL)
	{
		// Find Context
		context = pangolin::FindContext(name);
		if (context == NULL)
		{
			// Create new
			std::shared_ptr<pangolin::PangolinGl> newcontext(new pangolin::PangolinGl());
			AddNewContext(name, newcontext);
			context = newcontext.get();
			std::cout << "Pangolin Context" << name << "created." << std::endl;
		}
		else
		{
			std::cout << "Pangolin Context" << name << "already exists." << std::endl;
		}
		// Make Current Context
		context->MakeCurrent();

		//  Initialize GLEW
		glewExperimental = GL_TRUE; // GL_FALSE; 
		GLenum error = glGetError();

		if (error != GL_NO_ERROR)
		{
			std::cout << "OpenGL Error: " << error << std::endl;
		}

		GLenum glewinit = glewInit();
		if (glewinit != GLEW_OK) {
			std::cout << "Glew Error: " << glewGetErrorString(glewinit) << std::endl;
			exit(EXIT_FAILURE);
		}
	}
	else
	{
		std::cout << "Pangolin Current Context exists." << std::endl;
	}
}

#endif /* !__STATIC_GLEW_INIT_HPP__ */
```

hitsjt, 2017-02-14:
hey,have you solved this problem?or how to do the modification you mentioned?
Thank you!

spyd3rweb, 2017-02-16:
@hitsjt 
I woudn't call it a good solution, but it works.

Take the above code and save it as "static_glew_init.hpp" in your %GITHUB_DIR%\ElasticFusion\Core\src folder, and add it to the Headers filter under the efusion project in the libefusion solution.  

In "Shaders.h" add the include for "static_glew_init.hpp" and add a call to the staticGlewInit() function in the Shader constructor:

`#ifndef SHADERS_SHADERS_H_
#define SHADERS_SHADERS_H_

#include <pangolin/gl/glsl.h>
#include <memory>
#include "../Utils/Parse.h"
#include "../static_glew_init.hpp"
#include "Uniform.h"

class Shader : public pangolin::GlSlProgram
{
    public:
        Shader()
        {
		// glewInit
		staticGlewInit();
	}`

hys2015, 2017-06-20:
@spyd3rweb 
i did everything but i got these...

```
D:\GitForks\ElasticFusion\bin>ElasticFusion.exe
Creating live capture... success!
Waiting for first frame....................................................... got it!
Pangolin Current Context exists.
Pangolin Current Context exists.
Pangolin ContextMain created.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Error: out of memory: d:\gitforks\elasticfusion\core\src\cuda\containers\../convenience.cuh:68
```

hitsjt, 2017-06-20:
@hys2015 
look at this [https://github.com/mp3guy/ElasticFusion/issues/7](url)

hys2015, 2017-06-20:
@hitsjt 
it works, thank you!

iggyvolz, 2018-02-17:
@spyd3rweb What do you mean by "install" in Suite Sparse step 8?  Is there an installation step I need to do (I'm guessing this is comparable to a make install on linux)?  I don't have an include or lib64 directory in suitesparse so I'm guessing that's why

iggyvolz, 2018-02-17:
Oh I see, I had to explicitly build the "Install" project.

MarsTNT, 2018-05-11:
@hys2015 Hi, how could you solve the problem ---- MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.
MakeCurrent: Not available with non-pangolin window.
Pangolin ContextMain already exists.   ?

Could you explain more in detail please? Thank you in advance!

hys2015, 2018-05-16:
@MarsTNT lower this number https://github.com/mp3guy/ElasticFusion/issues/7

kcshum, 2019-08-08:
@hys2015 How much should I lower this number? I tried 768 but still got the error. What value do you use?

chuong, 2019-10-05:
@spyd3rweb I managed to successfully compile Core and GUI on Windows 10 using VS 2017. I also copied all DLL files to same folder of ElasticFusion.exe. However when running ElasticFusion.exe, I got Ordinal Not Found error:

> ---------------------------
> ElasticFusion.exe - Ordinal Not Found
> ---------------------------
> The ordinal 1681 could not be located in the dynamic link library D:\Users\ngu10t\workspace\ElasticFusion\GUI\build\Debug\efusion.dll. 
> ---------------------------
> OK   
> ---------------------------
> 

Any idea please?

omgitsraven, 2020-01-19:
I'm having the "glewInit already defined" error, but I don't understand how to follow your instructions:
```
In project properties for linker move efusion.lib below the include of glewd.lib 
```
Where is "project properties for linker"?

spyd3rweb, 2020-01-19:
@omgitsraven 
> Where is "project properties for linker"?
https://www.youtube.com/watch?v=or1dAmUO8k0&t=13m30s

omgitsraven, 2020-01-20:
In the Solution Explorer, the only Project that has a Linker section in its Properties is `ElasticFusion`, and it doesn't include `efusion.lib` or `glewd.lib`, only this:
```
ws2_32.lib
C:\Libraries\Pangolin-0.5\build\external\libjpeg\lib\jpeg.lib
C:\Libraries\Pangolin-0.5\build\external\zlib\lib\zlibd.lib
<different options>
```

MingwuZheng, 2020-03-15:
> @hys2015 Hi, how could you solve the problem ---- MakeCurrent: Not available with non-pangolin window.
> Pangolin ContextMain already exists.
> MakeCurrent: Not available with non-pangolin window.
> Pangolin ContextMain already exists. ?
> 
> Could you explain more in detail please? Thank you in advance!

> @MarsTNT lower this number #7

Hi, how could u solve this problem? I have the same problem but I cant find out where #7 is

MingwuZheng, 2020-03-15:
> @hys2015 Hi, how could you solve the problem ---- MakeCurrent: Not available with non-pangolin window.
> Pangolin ContextMain already exists.
> MakeCurrent: Not available with non-pangolin window.
> Pangolin ContextMain already exists. ?
> 
> Could you explain more in detail please? Thank you in advance!

have u solved it? I cant find any solution.

closed 2022-05-27 by mp3guy