mp3guy/ElasticFusion #71: Ground truth trajectory and estimated trajectory does not match (ICL-NUIM data set)

M1234Thomas, 2017-01-11:
Hello,

I have run ICL-NUIM living room data set using elastic fusion but the ground truth trajectory is not matching with the estimated trajectory. I tried to use the ground truth file along with '-p' argument and the reconstruction is erroneous. Anyone faced same issue? 

[image]

HTLife, 2017-01-12:
Same here, and I also want to fix this problem!

mp3guy, 2017-01-12:
You should try this; 

https://github.com/mp3guy/ElasticFusion/blob/master/GUI/src/MainController.cpp#L32

HTLife, 2017-01-14:
Did anyone know what is the scale factor of depth image (between ElasticFusion and ICL-NUIM)?

I convert ICL-NUIM "living_room_traj2_frei_png" to .klg by the following scaling factor (0.2).
```
cv::Mat depth = imread(strPath, cv::IMREAD_UNCHANGED);
double scale = 0.2;
depth.convertTo(depth, CV_16UC1, scale);
```
It seems that I'm getting closer to the right setting.
[image]

M1234Thomas, 2017-01-24:
@HTLife I have checked ur pngtoklg project. The dataset i had downloaded from ICL-NUIM has timestamp values as 1,2,3,4,5..etc. But your examples shows a different timestamp. Also the groundtruth trajectory also shows values 1,2,3,4 in http://www.doc.ic.ac.uk/~ahanda/VaFRIC/iclnuim.html.

HTLife, 2017-01-24:
@M1234Thomas 
ICL-NUIM use integer as timestamp.  The default format of pngtoklg is float number define by TUM RGB-D dataset.  You could load TUM format in ElasticFusion by -icl flag.

For example,
1. Convert png(ICL-NUIM) into klg with scale factor 0.0002
`./pngtoklg -w [path to living_room_traj2_frei_png directory] -o lrkt2_scale.klg -s 0.0002`
2. Load .klg into Elasticfusion
`./ElasticFusion -l lrkt2_scale.klg -icl`
3. Because the norm of ICL-NUIM is inverse, you could multiple -1 into 3th column of ground truth trajectory, which means y axis.
4. Compare trajectory generate by elasticfusion and ground truth trajectory (already times -1).
5. You should get the same result that I provided in previous post.

HTLife, 2017-01-24:
@M1234Thomas You should update the pngtoklg project for recent change.

M1234Thomas, 2017-01-25:
Thanks for the answer. I have tried multiplying -1 into the y axis. But the result looks like this
[image]

HTLife, 2017-01-25:
The tool script provided by TUM `evaluate_ate.py` cover the trajectory with red line which is hard to read.  I tried to use excel to draw the trajectory:
1. Download ICL-NUIM dataset `Living Room 'lr kt0'`
TUM RGB-D Compatible PNGs: 
[`living_room_traj0_frei_png`](http://www.doc.ic.ac.uk/~ahanda/living_room_traj0_frei_png.tar.gz)
2. `./pngtoklg -w '/media/rvl/data/dataset/icl/lrkt0/living_room_traj0_frei_png' -o traj0.klg -s 0.0002 -t`
3. `./ElasticFusion -l '/media/rvl/data/dataset/icl/lrkt0/living_room_traj0_frei_png/traj0.klg' `
4. edit the file `traj0.klg.freiburg` remove 0 at the front
5. Multiply -1 to y axis of ground truth file  `livingRoom0.gt.freiburg`
6. Draw by excel with x,y axis
[image]

The **shape** is similar, the difference between ground truth could be accumulated error of visual odometry.

xiaozhuka, 2017-03-21:
hello @HTLife sorry to brother you
do you know why the global model upside down and left right reversed when set -icl.
image can be found #82

HTLife, 2017-03-21:
@xiaozhuka 
ICL-NUIM has negative norm which is different from TUM.
If you are using my png_to_klg project to convert icl dataset, you could ignore this issue.  Because when use png_to_klg with icl parameter, it already revert it for you.  No need to set it in ElasticFusion again.

xiaozhuka, 2017-03-22:
@HTLife Got it, Thank you

Liuyangdong, 2017-03-24:
@HTLife 
Hi, is it correct to set both the RGB and the depth camera calibration matrix as follows when using ICL-NUIM dataset,
481.20,	0,	319.50
0,	480.00,	239.50
0,	0,	1
Thanks.

HTLife, 2017-03-25:
@Liuyangdong Correct

Liuyangdong, 2017-04-12:
@HTLife The value of fy in the camera intrinsic matrix should be 480.0 or -480.0 when using ICL-NUIM dataset?

HTLife, 2017-04-12:
Both -icl parameter in ElasticFusion and png_to_klg will times the inverted fy by -1.
So 480.0 should stay in positive.

Reference here: https://github.com/mp3guy/ElasticFusion/blob/31c874a2cd5edfadb377c05860e78bd34c2eda41/GUI/src/MainController.cpp#L305

Liuyangdong, 2017-04-17:
@HTLife Thank you for your kind reply!

ubuntuslave, 2017-06-19:
I suspect the GT poses given by ICL-NUIM obey a **Left Handed** coordinate system. Please, correct me if I'm wrong. Re-rendering the poses (as they were given) by POV-Ray produces the exact same camera trajectory that was originally rendered by ICL. Thus, I can think that these poses obey left handedness, but I can't be 100%. This is only my conclusion after trying to render new images in POV-Ray. I usually have to convert from "right-handed poses" to "left-handed", but this is not necessary here.

Any clues?

root29, 2018-04-04:
Hi , I have some problem about how to run ICL_NUM dataset.

./Elasticfusion -icl [path to living_room_traj0_frei_png] ?

foolboyly, 2018-05-10:
Hello!
I have two questions about evaluate trajectory:
1. the result of *.klg.freiburg is cameratrajectory or keyframetrajectory?
2.I use the TUM dataset to evaluate ate, the result is bad
[image]
but when i use -p groundtruth.txt, there was a error that n should equal 8(I already delete the first  three lines notes).the groundtruth.txt is:
1305031787.1206 0.4183 -0.4920 1.6849 -0.8156 0.0346 -0.0049 0.5775
1305031787.1306 0.4180 -0.4917 1.6854 -0.8137 0.0347 -0.0043 0.5802
1305031787.1406 0.4176 -0.4922 1.6860 -0.8133 0.0353 -0.0049 0.5807

HTLife, 2018-05-10:
@foolboyly groundtruth.txt should be comma separated (you could find the parsing code around that error message).

wine3603, 2018-07-20:
@HTLife Hi Jacky, I have 1 problem when using the ground truth file. 
when I run
` ./ElasticFusion -l /home/.../slam.klg  -p ~/modified_groundtruth.txt `
the gt.txt was modified from TUM's groundtruth.txt, timestamp value was
 multip with 1000 to meet  [EF -p mode setting](https://github.com/mp3guy/ElasticFusion/blob/master/GUI/src/Tools/GroundTruthOdometry.cpp#L42)
 `int` n = sscanf(line.c_str(), "%llu,%f,%f,%f,%f,%f,%f,%f", &utime, &x, &y, &z, &qx, &qy, &qz, &qw);`
 
I got this a fixed camera, the camera never moves.
I think this maybe the -p mode use different xyz coordinate system, 
I modified from right hand to left hand coordinate (x/y/z: right/down/forward) but 
still no luck.
I also tried to convert the transforms in gt.txt  from word space to camera space, 
still not work. 
I find other guy had the same problem with me:
[in stack overflow](https://stackoverflow.com/questions/45411489/elasticfusion-slam-algorithm-running-with-pose-graph-on-tum-rgb-d-benchmark)

HTLife, 2018-07-20:
@wine3603 Using debug mode to trace code first.  
The entry point could be 

1. GUI/src/Tools/GroundTruthOdometry.cpp    Line: 42
2. [MainController.cpp](https://github.com/mp3guy/ElasticFusion/blob/395c339b82e435801ba0225bf5062b94aa3b5af6/GUI/src/MainController.cpp#L262-L265) L262-L265

Viky397, 2021-02-16:
> @HTLife Hi Jacky, I have 1 problem when using the ground truth file.
> when I run
> `./ElasticFusion -l /home/.../slam.klg -p ~/modified_groundtruth.txt`
> the gt.txt was modified from TUM's groundtruth.txt, timestamp value was
> multip with 1000 to meet [EF -p mode setting](https://github.com/mp3guy/ElasticFusion/blob/master/GUI/src/Tools/GroundTruthOdometry.cpp#L42)
> `int` n = sscanf(line.c_str(), "%llu,%f,%f,%f,%f,%f,%f,%f", &utime, &x, &y, &z, &qx, &qy, &qz, &qw);`
> 
> I got this a fixed camera, the camera never moves.
> I think this maybe the -p mode use different xyz coordinate system,
> I modified from right hand to left hand coordinate (x/y/z: right/down/forward) but
> still no luck.
> I also tried to convert the transforms in gt.txt from word space to camera space,
> still not work. The result is:
> [image]
> 
> I find other guy had the same problem with me:
> [in stack overflow](https://stackoverflow.com/questions/45411489/elasticfusion-slam-algorithm-running-with-pose-graph-on-tum-rgb-d-benchmark)

Did you solve this error? I'm trying to do the same thing using the -p flag and the camera frame stays static. Thanks!

joshi-bharat, 2021-02-26:
If anyone is looking for the solution, @HTLife  use depth time stamp not the rbg ones.

closed 2022-05-27 by mp3guy

evertale888, 2024-07-29:
> Did anyone know what is the scale factor of depth image (between ElasticFusion and ICL-NUIM)?
> 
> I convert ICL-NUIM "living_room_traj2_frei_png" to .klg by the following scaling factor (0.2).
> 
> ```
> cv::Mat depth = imread(strPath, cv::IMREAD_UNCHANGED);
> double scale = 0.2;
> depth.convertTo(depth, CV_16UC1, scale);
> ```
> 
> It seems that I'm getting closer to the right setting. [image]
> 
> By the way, this is my conversion command `./pngtoklg -w [path to living_room_traj2_frei_png directory] -o lrkt2_scale.klg -s 0.0002` You could download pngtoklg project from (https://github.com/HTLife/png_to_klg)
> 
> ```
> ./ElasticFusion -l './lrkt2_scale.klg' -icl
> 
> ./evaluate_ate_dot.py './livingRoom2.gt_reverse_norm.freiburg' './lrkt2_scale.klg.freiburg' --plot './scale_Elas.png' 
> 
> Output: 0.111543
> ```

Do u know what the values for cam variable below for icl-nuim? Maybe your estimated value better than nothing.

1. Close/Far threshold. Baseline times.
2. Depth map values factor

i want to try run it in orbslam3 rgbd. The web doesnt mention these except for the cam calibration matrix.