mp3guy/ElasticFusion #75: How to adjust parameters to trigger "Global loop closure"?

HTLife, 2017-02-19:
Currently I'm working on applying TUM "fr2/360_kidnap" to ElasticFusion.  However, loop could not been detected.  (Kintinuous could detect loop but the map is broken.)

If I want to get a map with complete reconstruction result and loop closure, what would you guys suggest?

## Here are my experinment parameters and result
`./ElasticFusion -l '/dataset/tum/rgbd_dataset_freiburg2_360_kidnap/scale_0_0002.klg' -ftf -d 12 -c 4 -ft 0.1`
 - Do frame-to-frame RGB tracking.
 - Cutoff distance = 12m
 - Surfel confidence threshold = 4
 - Fern encoding threshold = 0.1

Final result screenshot:
[image]
Video: https://youtu.be/PWe2Cziccw8

I provide my .klg file here for result reproduce. (ftp://ccurvl.csie.io/slam_exp/)

I did both ElasticFusion and Kintinuous on TUM "fr2/360_kidnap"
  - ElasticFusion: [good] map reconstruction,  [bad] loop detection
  - Kintinuous: [good] loop detection, [bad] map reconstruction (maybe didn't find the right voxel threshold or other related...)

I also tried the instruction in https://github.com/mp3guy/ElasticFusion/issues/21 to adjust some parameters, but loop detection still don't work with TUM kidnap dataset.

mp3guy, 2017-02-19:
Please see Page 16 Table 2; http://www.thomaswhelan.ie/Whelan16ijrr.pdf

You would need to write new code to manage relocalisation and map merging if you want either of these methods to work on kidnapped sequences.

HTLife, 2017-02-19:
@mp3guy  Thank you for your quick response.
Before writing new code, I want to test the capability of place recognition module (both DBoW2 and Fern).
After trying out some fern related setting, loop still couldn't been correctly detected (Fern encoding threshold = 0.1).  Would that be the limitation of "fern" method?  Or I didn't tune it right?

closed 2017-02-27 by HTLife