mp3guy/ElasticFusion #77: OpenGL Error: XX (1280): GLSL Shader compilation failed: draw_global_surface.vert

anatoly-spb, 2017-03-01:
When I run the program build with Visual Studio 2013 on Windows 7 64bit I have problem with the draw_global_surface.vert shader:

```
Creating live capture... failed!
DeviceOpen using default: no devices found
GLSL Shader compilation failed: H:/tmp/ef/src/ElasticFusion/Core/src/Shaders/dra
w_global_surface.vert:
ERROR: 0:21: 'location' : syntax error parse error

GLSL Program link failed: Geometry shader object: no attached vertex shader obje
ct.

OpenGL Error: XX (1280)
In: H:\tmp\ef\src\Pangolin\include\pangolin/gl/gl.hpp, line 192
```

What is wrong?

heiwuya-heiwuya, 2020-09-28:
Hello, did you solved this problem? I met the same one. How did you solved it

LuoMing2, 2020-10-07:
I've met the same problem, this seems to can't select NVIDIA GPU, and I don't know how to solve it yet

closed 2022-05-27 by mp3guy