mp3guy/ElasticFusion #78: Run Error:  new Resize()

jiaxing-li, 2017-03-06:
When I run the ElasticFusion program build with Visual Studio 2013 on Windows 8.1 ,it crashed.

Run into this code when crashing.
GUI\src\MainController.cpp:
    resizeStream = new Resize(Resolution::getInstance().width(),
                              Resolution::getInstance().height(),
                              Resolution::getInstance().width() / 2,
                              Resolution::getInstance().height() / 2);
Crashes when calling the code for the AddPreprocessedShader function of the GlSlProgram class,
    if(!prog) {
        prog = glCreateProgram();
    }

Why does the glCreateProgram function crash?

@mp3guy

jiaxing-li, 2017-03-08:
The issues is solved.
Because there are other GLEW libraries in the computer, causing the library to conflict.
I build Pangolin and efusion  into a static library, so use no problem.

closed 2017-03-08 by jiaxing-li

hughes1217, 2020-04-09:
@jiaxing-li I also encountered the same problem, and used your solution to build pangolin and eFusion into a static library, but when building ElasticFusion, there was a link error, unable to resolve the external symbols. Do you know why?

错误	2	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) int __cdecl gettimeofday(struct timeval *,struct timezone *)" (__imp_?gettimeofday@@YAHPEAUtimeval@@PEAUtimezone@@@Z)，该符号在函数 "private: void __cdecl MainController::run(void)" (?run@MainController@@AEAAXXZ) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	3	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) void * __cdecl mempcpy(void *,void const *,unsigned __int64)" (__imp_?mempcpy@@YAPEAXPEAXPEBX_K@Z)，该符号在函数 "private: char * __cdecl Stopwatch::serialiseTimings(int &)" (?serialiseTimings@Stopwatch@@AEAAPEADAEAH@Z) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	4	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) public: __cdecl GPUTexture::GPUTexture(int,int,unsigned int,unsigned int,unsigned int,bool,bool)" (__imp_??0GPUTexture@@QEAA@HHIII_N0@Z)，该符号在函数 "public: __cdecl GUI::GUI(bool,bool)" (??0GUI@@QEAA@_N0@Z) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	5	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) public: static class Resolution const & __cdecl Resolution::getInstance(int,int)" (__imp_?getInstance@Resolution@@SAAEBV1@HH@Z)，该符号在函数 "public: __cdecl MainController::MainController(int,char * * const)" (??0MainController@@QEAA@HQEAPEAD@Z) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	6	error LNK2001: 无法解析的外部符号 "__declspec(dllimport) public: static class Resolution const & __cdecl Resolution::getInstance(int,int)" (__imp_?getInstance@Resolution@@SAAEBV1@HH@Z)	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\LiveLogReader.obj
错误	7	error LNK2001: 无法解析的外部符号 "__declspec(dllimport) public: static class Resolution const & __cdecl Resolution::getInstance(int,int)" (__imp_?getInstance@Resolution@@SAAEBV1@HH@Z)	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\RawLogReader.obj
错误	8	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) public: static class Intrinsics const & __cdecl Intrinsics::getInstance(float,float,float,float)" (__imp_?getInstance@Intrinsics@@SAAEBV1@MMMM@Z)，该符号在函数 "public: __cdecl MainController::MainController(int,char * * const)" (??0MainController@@QEAA@HQEAPEAD@Z) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	9	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) public: static class Parse const & __cdecl Parse::get(void)" (__imp_?get@Parse@@SAAEBV1@XZ)，该符号在函数 "class std::shared_ptr<class Shader> __cdecl loadProgramFromFile(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?loadProgramFromFile@@YA?AV?$shared_ptr@VShader@@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@2@00@Z) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	10	error LNK2001: 无法解析的外部符号 "__declspec(dllimport) public: static class Parse const & __cdecl Parse::get(void)" (__imp_?get@Parse@@SAAEBV1@XZ)	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\LiveLogReader.obj
错误	11	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) public: int __cdecl Parse::arg(int,char * *,char const *,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)const " (__imp_?arg@Parse@@QEBAHHPEAPEADPEBDAEAV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)，该符号在函数 "public: __cdecl MainController::MainController(int,char * * const)" (??0MainController@@QEAA@HQEAPEAD@Z) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	12	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) public: int __cdecl Parse::arg(int,char * *,char const *,float &)const " (__imp_?arg@Parse@@QEBAHHPEAPEADPEBDAEAM@Z)，该符号在函数 "public: __cdecl MainController::MainController(int,char * * const)" (??0MainController@@QEAA@HQEAPEAD@Z) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	13	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) public: int __cdecl Parse::arg(int,char * *,char const *,int &)const " (__imp_?arg@Parse@@QEBAHHPEAPEADPEBDAEAH@Z)，该符号在函数 "public: __cdecl MainController::MainController(int,char * * const)" (??0MainController@@QEAA@HQEAPEAD@Z) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\MainController.obj
错误	14	error LNK2019: 无法解析的外部符号 "__declspec(dllimport) public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl Parse::shaderDir(void)const " (__imp_?shaderDir@Parse@@QEBA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)，该符号在函数 "class std::shared_ptr<class Shader> __cdecl loadProgramFromFile(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?loadProgramFromFile@@YA?AV?$shared_ptr@VShader@@@std@@AEBV?$basic_string@DU?$ch [... cut ...] raits@D@std@@V?$allocator@D@2@@std@@XZ)，该符号在函数 "public: virtual class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const __cdecl LiveLogReader::getFile(void)" (?getFile@LiveLogReader@@UEAA?BV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) 中被引用	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\LiveLogReader.obj
错误	58	error LNK1120: 53 个无法解析的外部命令	E:\project\ElasticFusion\ElasticFusion-master\GUI\src\build64\Release\ElasticFusion.exe

hughes1217, 2020-04-11:
The reason was found. It was not deleted "__declspec(dllimport)". Your solution works, thank you!