mp3guy/ElasticFusion #81: Need you advise , what hw devices do ElasticFusion support now , which one is the accurate one?

xj1982xj, 2017-03-21:


xiaozhuka, 2017-03-21:
ElasticFusion does support RealSense, but may be not the accurate one

xj1982xj, 2017-03-21:
Hi xiaozhuka , very thanks. two more question want to consult : 

1. which device do you think is the most accurate one for ElasticFusion , I saw "ElasticFusion Real-time Light Source Detection" video in youtube channel , do you know which hw device this video is using ?    

2. can ElasticFusion has the capability to export the scanned environment as an obj or fbx for other 3d modeling software to use like 3dsmax or maya ?   thanks.

DanbingZhu, 2017-04-21:
@xiaozhuka Hi, is it possible for you to briefly list the steps required to get elastic fusion working with realsense?

ShashankUppoor, 2017-04-21:
@xiaozhuka 
Does ElasticFusion with Realsense camera work only on Windows?

zhangduoshou, 2017-06-22:
@xiaozhuka I have the same question with DanbingZhu

whubaichuan, 2019-06-24:
> @xiaozhuka I have the same question with DanbingZhu

Hi, do you solve the problem?

closed 2022-05-27 by mp3guy