mp3guy/ElasticFusion #83: Possible issue with initICP in RGBDOdometry.cpp

kzampog, 2017-04-06:
Hi,

Thanks for releasing this code!

I was looking at the code in RGBDOdometry.cpp and noticed that calls in lines 162-163 have the same arguments:

        resizeVMap(vmaps_curr_[i - 1], vmaps_curr_[i]);
        resizeNMap(vmaps_curr_[i - 1], vmaps_curr_[i]);

while it seems like they should read:

        resizeVMap(vmaps_curr_[i - 1], vmaps_curr_[i]);
        resizeNMap(nmaps_curr_[i - 1], nmaps_curr_[i]);

Of course, I may be missing something. This particular variant of initICP() is used in line 487 of ElasticFusion.cpp.

Also, I was wondering if you were planning to release RGBDOdometry as a separate package (in the spirit of ICPCUDA which it seems to build upon).

Thank you!

Best,
Kostas

mp3guy, 2017-04-07:
Nice, great find! That's a pretty significant bug that was probably making the geometric term go to crap below the first pyramid level. I'd imagine the photometric term was doing all of the heavy lifting below that. 

I was planning to but never got the time, hopefully in the future.

closed 2017-04-07 by mp3guy

kzampog, 2017-04-25:
Hi again!

I have another question about RGBDOdometry. The version of `initICP` that directly takes a depth map does not seem to update the `vmaps_tmp` variable which is later used by `initRGB`, so the new depth map (`nextDepth[0]`) generated in line 212 seems to be based on the 'model' geometry.

I'm not sure if this is intended, but it seems from the code that the two `initICP` variants behave differently regarding this.

Thanks again!

Best,
Kostas

mp3guy, 2017-04-26:
Hmm, interesting find. It's a very long time since I wrote this code, but if I test what looks like a fix, performance on a benchmarks is significantly worse. So as you observed it's non-obvious as to what's going on here, but the way it is now is the way it works. It may be the case that something is poorly named. Long on my to do list is a big refactor and clean up of the tracking code (including visualising residuals etc...), but finding the time for that is difficult.

kzampog, 2017-04-26:
Cool, thanks a lot! I'll keep you updated if I happen to gain any insight on this.

commit 31c874a2 (2017-04-07): Fix normal map pyramid bug, closes #83