mp3guy/ElasticFusion #84: installation build error

darthvader2512, 2017-04-11:
this is what I am getting when I try to build elastic fusion

Severity	Code	Description	Project	File	Line	Suppression State
Error	C2440	'<function-style-cast>': cannot convert from 'initializer list' to 'Uniform'	efusion	C:\Users\vikra\Downloads\Project\ElasticFusion-master\Core\src\GlobalModel.cpp	265	
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2440	'<function-style-cast>': cannot convert from 'initializer list' to 'Uniform'	efusion	C:\Users\vikra\Downloads\Project\ElasticFusion-master\Core\src\Shaders\FeedbackBuffer.cpp	147	
Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3073	The command "setlocal
"C:\Program Files\CMake\bin\cmake.exe" -DBUILD_TYPE=Debug -P cmake_install.cmake
if %errorlevel% neq 0 goto :cmEnd
:cmEnd
endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone
:cmErrorLevel
exit /b %1
:cmDone
if %errorlevel% neq 0 goto :VCEnd
:VCEnd" exited with code 1.	INSTALL	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets	133

Nikshukech, 2017-07-18:
@darthvader2512 have you decided this problem?

kemangjaka, 2019-02-06:
Hi, I also had the same problem but solved by compiling Pangolin again with Eigen.

omgitsraven, 2020-01-03:
How did you compile Pangolin with Eigen?

https://github.com/stevenlovegrove/Pangolin/issues/355 seems to suggest I can give cmake an `EIGEN_INCLUDE_DIR` value pointing to the Eigen sources, but doing so (and rebuilding both Pangolin and ElasticFusion) didn't help…

closed 2022-05-27 by mp3guy