mp3guy/ElasticFusion #86: ElasticFusion.exe crashing

darthvader2512, 2017-04-16:
After executing the C:\Program Files\ElasticFusion\bin\ElasticFusion.exe file, the program (main window) crashes. However the command prompt shows no errors. Given are the two images of the DLLs used in the same folder; as well as the command prompt with the main window while it crashes. I am working on a project currently which requires elastic fusion and can use some immediate help with this. Really appreciate any help. Thanks in advance!

[image]
[image]

darthvader2512, 2017-04-16:
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\ElasticFusion.exe'. Module was built without symbols.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\zlib.dll'. Module was built without symbols.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\windows.storage.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\powrprof.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\profapi.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\OpenNI2.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\opengl32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\efusion.dll'. Module was built without symbols.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\msvcr100.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\glu32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\msvcp140.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\liblapack.dll'. Module was built without symbols.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\libblas.dll'. Module was built without symbols.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\libgcc_s_sjlj-1.dll'. Module was built without symbols.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\libgfortran-3.dll'. Module was built without symbols.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\ElasticFusion\bin\libquadmath-0.dll'. Module was built without symbols.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\uxtheme.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\msctf.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\oleaut32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\Nahimic\Nahimic2\UserInterface\x64\Nahimic2DevProps.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Program Files\Nahimic\Nahimic2\UserInterface\x64\Nahimic2OSD.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\dwmapi.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\nvoglv64.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\setupapi.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\wtsapi32.dll'. Cannot find or open the PDB file.
[... cut ...]
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\nvcuda.dll'. Cannot find or open the PDB file.
'ElasticFusion.exe' (Win32): Loaded 'C:\Windows\System32\nvfatbinaryLoader.dll'. Cannot find or open the PDB file.
The thread 0x6a8 has exited with code 0 (0x0).

Exception thrown at 0x0000000000000000 in ElasticFusion.exe: 0xC0000005: Access violation executing location 0x0000000000000000.
Unhandled exception at 0x0000000000000000 in ElasticFusion.exe: 0xC0000005: Access violation executing location 0x0000000000000000.

The program '[10900] ElasticFusion.exe' has exited with code 0 (0x0).

zhangxaochen, 2017-06-11:
You may first run the program in debugging mode, break where the **access violation** issue happens, and paste the call stack here.

A possible solution relates to your issue may be the "static_glew_init.hpp" patch by @spyd3rweb: https://github.com/mp3guy/ElasticFusion/issues/70