mp3guy/ElasticFusion #89: Use ground truth pose in ROS gazebo

againxx, 2017-04-24:
Hi, I'am now using ElasticFusion in gazebo simulation environment. I use ROS topic to provide ElasticFusion with RGB-D data and use odom data as ground truth position. But it seems that the reconstructed objects did not match very. There are two chairs with a little distance between them, but there should be one. What should I do to correctly use ground truth position? I just put it to currentPose before processFrame.

zdczdcc, 2017-08-14:
Hi,@againxx,
Have you solved the problem yet?

againxx, 2017-08-14:
Yes, It took me a lot of time to combine ElasticFusion with ROS and gazebo.

againxx@mail.ustc.edu.cn
 
From: zdczdcc
Date: 2017-08-14 18:03
To: mp3guy/ElasticFusion
CC: againxx; Mention
Subject: Re: [mp3guy/ElasticFusion] Use ground truth pose in ROS gazebo (#89)
Hi,@againxx,
Have you solved the problem yet?

zdczdcc, 2017-08-14:
Hi,@againxx ,
I've found that "groundTruthOdometry->getTransformation()" didn't get the true pose reading from the dataset in the origin source code.Could you give me some help to modify it?

closed 2017-11-14 by againxx