mp3guy/ElasticFusion #90: Poorer map quality using Intel Realsense R200

henrywarhurst, 2017-04-26:
Has anyone had the issue of poorer map quality when using the Intel Realsense R200 versus when using the ASUS Xtion Pro Live? I am able to build much better maps of the same environments using the Xtion Pro Live than the RS R200. Is the RS R200 simply an inferior camera for this task or am I doing something wrong?

@fsrajer I noticed you were the contributor for the Realsense support commit - can you provide any insights on this issue?

Many Thanks,
Henry

mp3guy, 2017-04-26:
I think if you compare them side by side, the R200 is an inferior sensor in terms of precision and accuracy.

fsrajer, 2017-04-26:
Yes, it is definitely because of the quality of the sensor. I actually abandoned the idea of using R200 because I just couldn't get the required quality of the reconstruction.

henrywarhurst, 2017-04-27:
Thanks for your insights @mp3guy, @fsrajer. Closing this out.

closed 2017-04-27 by henrywarhurst