mp3guy/ElasticFusion #91: Segmentation fault (core dumped)

Amranbhuiyan, 2017-05-02:
@mp3guy I follow the exact instructions and running it on Ubuntu 14.04 , still receiving the following running issue. Could you please help me to figure out this problem:

amran@amran:~/ElasticFusion/GUI/build$ ./ElasticFusion -l dyson_lab.klg
Segmentation fault (core dumped)

Can anyone help me to figure out this problem?

mp3guy, 2017-05-02:
Compile it with -g and run it through gdb to find out what line it's segfaulting on.

Amranbhuiyan, 2017-05-03:
Can you please please look at it?  and I am using GTX 1070 which I think support as it speeds is 5.7 TFLOPS    

Problem is here:
frameToModel.initFirstRGB(textures[GPUTexture::RGB]);

Amranbhuiyan, 2017-05-12:
@mp3guy I found the problem. Actually, the GPU I was using in my system was not getting power to run this algorithm.

Thanks for your help.

SunshineDou, 2018-11-18:
Hello, I also run into this problem these days. when I run with the "dyson_lab.klg" dataset, It can run normally. but I want to run it with my own dataset, as I am doing a project these months, I get the images by Realsense and converted it to klg format, But when I run it in ElasticFusion, It failed, the outcome is:
Segmentation fault (core dumped)

I want to figure out the problem, what's the problem? should I compile it with -g to find out the problem?
and I want to know how to compile it with -g, I have tried several times, but failed, could you please give a more detailed discription. Thank you very much!

closed 2022-05-27 by mp3guy

tfy14esa, 2022-07-02:
> @mp3guy I found the problem. Actually, the GPU I was using in my system was not getting power to run this algorithm.
> 
> Thanks for your help.

@Amranbhuiyan can you elaborate what you mean by "GPU not getting power"? I am facing the same issue as you, but have not yet concluded what is the cause.