mp3guy/ElasticFusion #96: GLSL 3.30 is not supported

archenroot, 2017-06-17:
```
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/draw_global_surface.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/draw_global_surface.geom:
;
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/draw_global_surface_phong.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/quad.geom:
, 0.0, 1.0);
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/fxaa.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

OpenGL Error: XX (1280)
In: /tmp/portage/media-libs/pangolin-9999/work/pangolin-9999/include/pangolin/gl/gl.hpp, line 192
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/quad.geom:
, 0.0, 1.0);
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/resize.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/quad.geom:
, 0.0, 1.0);
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/resize.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/quad.geom:
, 0.0, 1.0);
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/resize.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

OpenGL Error: XX (1280)
In: /usr/include/pangolin/gl/gl.hpp, line 562
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/index_map.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/index_map.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/quad.geom:
, 0.0, 1.0);
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/visualise_textures.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

OpenGL Error: XX (1280)
In: /usr/include/pangolin/gl/gl.hpp, line 192
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/splat.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/depth_splat.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/splat.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/combo_splat.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/init_unstable.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/draw_feedback.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/draw_feedback.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/draw_global_surface.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/draw_global_surface.geom:
;
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/draw_global_surface.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/data.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/data.geom:
osition;
GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/data.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/update.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/copy_unstable.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

GLSL Shader compilation failed: /home/zangetsu/proj/3d-frameworks/ElasticFusion/Core/src/Shaders/copy_unstable.geom:

OpenGL Error: XX (1280)
In: /usr/include/pangolin/gl/gl.hpp, line 192
Neoprávněný přístup do paměti (SIGSEGV)

```

archenroot, 2017-06-19:
This error is related to case when I have CPU Intel and also Nvidia graphics card on my laptop. I forgot that the by default I use Intel (to save battery), so I installed Bumblebee and run this via optirun on discrete Nvidia card and that has resolved the issue.

closed 2017-06-19 by archenroot

yyxr75, 2018-11-21:
Hi sir,
I'm sorry to bother you, I was testing the ElasticFusion DEMO and trying to make it through with Realsense D415. Luckily the debug message of findong out device showed out, but then the errors came up:

> Creating live capture... failed!
DeviceOpen using default: no devices found
Creating live capture... start
Intel RealSense D415 746112060940
Auto Exposure enable
Auto White Balance disable
success!
Waiting for first frame.......... got it!
 GLSL Shader compilation failed: /home/yang/Desktop/ElasticFusion/Core/src/Shaders/draw_global_surface.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

You mentioned the reason of it, but I still have no idea how to do with them, would you mind to give me the specific method to solve them.

I'm on the ubuntu-16.04

Sincerely
Best wishes
yang

yyxr75, 2018-11-27:
Yes, after i reinstall the nvidia driver and cuda toolkit, this error never show up again. Thank you so much.

LittleLampChen, 2019-04-27:
@archenroot @yyxr75 @dcanelhas I also came across with this problem with 4 GPU TITAN Card. Can it work on multi GPU card?

heiwuya-heiwuya, 2020-09-09:
Hello , I just one card Nvidia GTX1660Ti.  it shows no intel card on Ubuntu . when i ran elasticfusion on Ubuntu16.04, it shows also OpenGL Error: XX (1280)