mp3guy/ElasticFusion #101: (realsense) DeviceOpen using default: no devices found

atenpas, 2017-07-11:
I always get this error on Ubuntu 14.04 using an intel realsense sr300: 

> DeviceOpen using default: no devices found

I can access the realsense without issues through the examples provided by librealsense.

zhangxiaoxuan1, 2017-09-23:
I use R200 and encounter the same issue. On readme it says "Essentially by default ./ElasticFusion will try run off an attached ASUS sensor live.", so I guess by default it only tries to find an ASUS sensor? I didn't read the code so not sure

iliis, 2018-03-02:
The RealSense SDK is only included in Windows by default. I had to change a few things to get it running on Linux:

- remove the `if(WIN32)` guards in `GUI/src/CMakeLists.txt` (line 7 and line 30)
- fix `GUI/src/FindRealSense.cmake` so that your librealsense installation can be found: Possibly adapt the `REALSENSE_ROOT` variable (I changed it from `/usr/local` to `/usr`) and rename `[lib]REALSENSE` to `[lib]realsense` in the `FIND_PATH`/`FIND_LIBRARY` calls.

After a full recompilation (delete your build folder!) it worked with my ZR300 (turns out my GPU is too slow :( )
If you installed librealsense from the repositories make sure you have the headers as well (install `librealsense-all-dev` on Ubuntu).

RajathSwaroop, 2018-03-22:
Hi Iliis, I was curious to know if the procedure you mentioned above works for intel realsense D415? I have installed SDK for realsense (SDK 2.10). Is this same as librealsense or is it something different. I am new to realsense so I dont know if they mean the same.

iliis, 2018-03-23:
No, I don't think the SDK and librealsense are the same thing. However, I haven't used the SDK at all, so I don't know for sure. Also, my ZR300 uses the legacy librealsense branch, as it isn't supported by the newest versions anymore, so I can't help you with D415 specific questions either, I'm sorry.
But in general: Try to get librealsense to work (there are some examples that should work, like cpp-capture). Afterwards it's quite easy to integrate with ElasticFusion. However, I also found that at least the ZR300 is much worse than the ASUS XtionPro when it comes to creating depth data. Also, ElasticFusion seems to have been tuned to the XtionPro, I had to fiddle with some parameters in the algorithm to get things like loop closure working at all.

tobycollins, 2018-06-14:
Are there any plans to support the new librealsense sdk (v2)?

benlee12, 2018-07-18:
seems can support old realsense cameras, cant support sdk2.0 cameras(sr300 and d400). hope can support them someday

whubaichuan, 2019-06-24:
@RajathSwaroop Hi, can you make it work with realsenseSDK2 ?

leop123, 2019-09-25:
I also have a trouble with this. 

> Creating live capture... failed!
DeviceOpen using default: no devices found
X11 Error: GLXBadFBConfig
Pangolin X11: Indirect GLX rendering context obtained
Segmentation fault (core dumped)

I'm using RealSense D435i.

liu515, 2020-07-23:
> I also have a trouble with this.
> 
> > Creating live capture... failed!
> > DeviceOpen using default: no devices found
> > X11 Error: GLXBadFBConfig
> > Pangolin X11: Indirect GLX rendering context obtained
> > Segmentation fault (core dumped)
> 
> I'm using RealSense D435i.
I'm using RealSense D435i,too.how do you solve it?

ZhenghaoL, 2020-08-05:
@liu515 Hello, i'm using Realsense D435, with the help of #161 i ran the live mode successfully. But still a problem exits, that the save function don't work at all, i cant get the ply file. Hope you can solve it.

a333klm, 2020-08-07:
@ZhenghaoL what does not work? Does it crash? Or is there simply no file?

ZhenghaoL, 2020-08-07:
@a333klm Hi, simply no file. It is solved, i found it in root dic as a live.ply. Thanks for reply.

closed 2022-05-27 by mp3guy