mp3guy/ElasticFusion #105: Running ElasticFusion with .freiburg file

Mirnail, 2017-07-27:
I run the EF on **rgbd_dataset_freiburg1_desk** . I retrieved the ".freiburg"  trajectory file. I a bit  changed it to the format that EF expects to run it with "-p" flag. I expected to get same construction because i am running the exact same trajectory output from previous run. But it seems that i missing something. I am getting totally distorted graph.

I just changed timestamp from seconds to microsenconds by multiplying it to 1000000. 

`/path_to_EF/./ElasticFusion -l /path_to_data/rgbd_dataset_freiburg1_desk/test2.klg -p /path_to_data/rgbd_dataset_freiburg1_desk/modified_freiburg.txt`

Can you tell me what i am missing? Where could be my mistake?

Mirnail, 2017-08-03:
Solved by finding transformation matrix .

closed 2017-08-03 by Mirnail

AkshayHinduja, 2017-09-18:
Hi @Mirmix could you tell me what changes you made the pose file? I am trying to do the same.

Mirmix, 2017-09-18:
| 0 0 1|
|-1 0 0|
|0 -1 0|

You should apply transformation matrix to both rotation and translation

AkshayHinduja, 2017-09-19:
@Mirmix weirdly though I always fall into the assert n==8 break point and even if I comment that out (in Groundtruthodometry.cpp) the camera pose remains static, with surfels being drawn on the same patch continuosly. Did you make changes only in groundtruthodometry.cpp?

Mirmix, 2017-09-20:
Is your input file in this format 

`          "%llu,%f,%f,%f,%f,%f,%f,%f"`

It should be in above format probably it is a reason of assert failure.

AkshayHinduja, 2017-09-20:
Yes, I ensured that I am saving in the same format as its reading in by typecasting in the llu/f formats accordingly. Still it falls into the assertion. It shows n = 1, which I believe means it doe snot read beyond the timestamp perhaps?

Mirmix, 2017-09-22:
Can you copy and paste your file here. At least first few lines. And is your timestamp information same with inside the klg file.

AkshayHinduja, 2017-09-22:
I figured out the error I was making in reading the lines, and that is working fine now. For the transformation matrix, are you applying this transformation to the vector t and quaternion q (after converting to a rotation matrix) in  GroundTruthOdometry::loadTrajectory() and then pushing it to the isometry T?  
My reconstruction is still a bit messy on doing that though.

Mirmix, 2017-09-25:
You can apply tramsformation matrix directly on quaternion. 
```
| 0 0 1 0|   |    Rx    |
|-1 0 0 0|   |    Ry    |
|0 -1 0 0| * |   Rz     |
|0  0 0 1|    |   Rw    |
```

AkshayHinduja, 2017-09-25:
Got it, thank you!

Mirmix, 2017-09-26:
It was my pleasure.

Mirmix, 2017-09-26:
It was my pleasure :)

Viky397, 2021-02-09:
Hello, I am trying to pass in my own groundtruth pose file and am running into the n==8 assertion error. Could you please let me know what you did to fix this error? :) @AkshayHinduja

tfy14esa, 2022-08-03:
> I figured out the error I was making in reading the lines, and that is working fine now. For the transformation matrix, are you applying this transformation to the vector t and quaternion q (after [...]

I think I am facing the same issue as you had. What changes did you make when reading the file? Did you alter the source code?