mp3guy/ElasticFusion #109: run with ground truth pose

zdczdcc, 2017-08-14:
I'm going to run the ElasticFusion with ground truth pose given by the file livingRoom2.gt.freiburg from the ICL-NUIM.And i've convert the origin depth and rgb data to klg file by `png_to_klg`.
When i run with the command ./ElasticFusion -l livingRoom2.klg,it was ok.
But it was error(the reconstruction result was erroneous visibly) when i run with ./ElasticFusion -l livingRoom2.klg -p livingRoom2.gt.freiburg.
What should i do to run correct with the `-p` option?Thanks a lot!

mp3guy, 2017-08-14:
The format used for ground truth poses is not that of freiburg's, if you look at https://github.com/mp3guy/ElasticFusion/blob/master/GUI/src/Tools/GroundTruthOdometry.cpp you can see what format it expects.

closed 2017-08-14 by mp3guy

zdczdcc, 2017-08-15:
@mp3guy ,
thanks for your reply.I've been known about that.