mp3guy/ElasticFusion #114: Pangolin Error

mavbot66, 2017-09-12:
After running the bash file and building everything on freshly installed Ubuntu 16.04 LTS , I connect the KInect XBox 360 and run the ElasticFusion exe under GUI/build folder.I get this following error 
```
DeviceOpen using default: no devices found
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Unable to retrieve framebuffer options
Aborted (core dumped)
```
Some doubts are
1.) Pangolin is supposed to be build last but appears quite above in the bash file
2.) The NiViewer also shows "DeviceOpen using default: no devices found"
3.) I have nouveau.modest = 0 set in the grub for some ubutnu installation issue. Can this effect the output?
4.) Shall I try to make OpenNI work on a freshly installed Ubuntu then try for the bash file?Will the OpenNI in the bash file conflict with a already installed working OpenNI?

shanjichao, 2018-04-05:
@mavbot66 hi,Did you solve that problem?
Creating live capture... failed!
DeviceOpen using default: no devices found

heruiyuan, 2020-12-07:
@shanjichao I have the same problem. Have you solved it?

closed 2022-05-27 by mp3guy