mp3guy/ElasticFusion #117: zqemm

rnunziata, 2017-10-07:


rnunziata, 2017-10-07:
```
/ElasticFusion/GPUTest/build$ make
[ 50%] Linking CXX executable GPUTest
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `zgemm_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `zherk_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `zgemv_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `zpotrf_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `ztrsm_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `amd_realloc'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `dgemv_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `dgemm_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `dtrsm_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `amd_malloc'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `amd_printf'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `ztrsv_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `amd_free'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `dsyrk_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `amd_calloc'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `dpotrf_'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `amd_2'
/home/rjn/opencv3-p3-code/tracking_and_slam/newstuff/ElasticFusion/GPUTest/src/../../Core/build/libefusion.so: undefined reference to `dtrsv_'
collect2: error: ld returned 1 exit status

```

ChenjieWang, 2020-02-09:
Hello, I also encountered the same problem, did you solve it?

closed 2022-05-27 by mp3guy