mp3guy/ElasticFusion #118: Did ElasticFusion run correctly on my computer? (i7, Nvdia GT 1080)

lhanaf, 2017-10-20:
I just downloaded the code and compiled it on ubuntu 16.04, and run it on dyson_lab.klg data. The reconstructed results seems to be wrong, later part of the map got a wrong orientation like this picture:

[image]

mp3guy, 2017-10-20:
I can't reproduce this unfortunately.

lhanaf, 2017-10-21:
I just noticed that I'm using a wrong parameter for EFusion. Now I'm using the **default parameters** and get a map seems nice:

[image]

Great work! However, there is still some in-consistences like monitors appeared multi-times. Can the results be better by choosing better parameters?

Thanks a lot for your help.

Best Regards

mp3guy, 2017-10-21:
I just checked to see if there was a regression, but this doesn't happen for me.

nonlinear1, 2018-04-11:
Me, too ! I use ElasticFusion with kinect2 to map my office. The same problem occures. I think there is something wrong with loop closure ?

xingxing4182, 2019-02-16:
> Me, too ! I use ElasticFusion with kinect2 to map my office. The same problem occures. I think there is something wrong with loop closure ?

> Me, too ! I use ElasticFusion with kinect2 to map my office. The same problem occures. I think there is something wrong with loop closure ?

hi，when you use ElasticFusion with kinect2 to map your office the terminal output strs << "Your GPU \"" << dev << "\" isn't in the ICP Step performance database, please add it"?

closed 2022-05-27 by mp3guy