mp3guy/ElasticFusion #121: Can we run ElasticFusion offline from the depth and rgb maps.

mavbot66, 2017-11-10:
Hi..after seeing the dyson_lab.klg file run offline by ElasticFusion , I am just curious to know is it possible to run ElasticFusion offline just from a series of captured depth maps and rgb images. If yes how to go about that ? Any procedure by which we can convert our pre captured depthmaps into a .klg file and run it?

martinruenz, 2017-11-10:
You could use this tool to convert images and depth maps to klg files:
https://github.com/martinruenz/dataset-tools/blob/master/convert_imagesToKlg/main.cpp

mavbot66, 2017-11-21:
I am able to successfully run Logger2 by mp3guy. As per your link do I need to copy your link code into Logger2.h file of mp3guy ?Kindly please clarify how to use your code. I

martinruenz, 2017-11-21:
The easiest way would be to build the CMake file in the root of this repository, which also builds `convert_imagesToKlg`. You do not require Logger2.h for that.

mavbot66, 2017-11-21:
Some doubts ...sorry I am completely a MATLAB guy !!!
1.) On the page of Logger2 there is this line "Frames can be cached in memory and written out when logging is finished, or streamed to disk". The GUI of Logger2 does not give any such option to save the depth maps along with data logging or is it I am missing out something.
2.) When you ask me to build from the cmake file do you mean the cmake file of your project dataset-tools? or else how will I build convert_imagesToKlg. 

"

martinruenz, 2017-11-21:
1. Logger2 is recording rgb and depth frames, (all) the data is stored in the binary klg format. So if you want to record your own sequences, you should use this tool and ElasticFusion.
2. I was referring to this file: https://github.com/martinruenz/dataset-tools/blob/master/CMakeLists.txt which builds all the dataset tools.

mavbot66, 2017-11-21:
Actually I want to access the all the depth maps which ElasticFusion uses to build the final 3D point cloud. Somehow I found it not possible so I planned to record depth maps and build the 3D point cloud offline. I can do that with Logger2 -> klg -> ElasticFusion. But with Logger2 as you mentioned the depthmaps already gets compressed and packed in binary klg format. So can I convert my series of recorded depthmaps to klg by using your convert_imagesToKlg ?

HTLife, 2017-11-22:
@mavbot66 
You might be interested in this project https://github.com/HTLife/png_to_klg

closed 2017-11-22 by mavbot66