mp3guy/ElasticFusion #122: Extract depthmap from klg file

3dguy1, 2017-11-22:
Hi...I have been successful in running ElasticFusion online and now intend to run it offline. I am able to run the .klg file too. However when I run the .klg file I see in the GUI of ElasticFusion the depthmap being displayed. Can we extract depth map from the .klg file? If yes then how?

HTLife, 2017-11-22:
Press save button on GUI, it will save map to ply format

3dguy1, 2017-11-22:
But save button saves the reconstructed 3D structure in ply format. I want the depth maps used to build that 3D structure.

HTLife, 2017-11-22:
https://github.com/martinruenz/dataset-tools/blob/master/convert_imagesToKlg/main.cpp

The code provided above is how klg is been made.
You could invert the process to get depth rub and d images.

Why would you like to do so?
There are many rgbd dataset likes TUM and ICL-NIUM.

3dguy1, 2017-11-22:
I need to use/access the depth maps which ElasticFusion uses to create the 3D structure of my own scene.

martinruenz, 2017-11-22:
https://github.com/martinruenz/dataset-tools/blob/master/convert_klg/main.cpp
Extracts klg files.

mavbot66, 2017-11-23:
convert_klg worked like a charm....   ./convert_klg -i "..."  -o "..."   -a -depthpng

closed 2017-11-23 by 3dguy1

Viky397, 2021-01-19:
Hello! When I run ElasticFusion with my own .klg, I do not see my depth images being displayed in the bottom row of the GUI. Any help would be appreciated, thank you!