mp3guy/ElasticFusion #124: GPUTest not work

andjsmile, 2017-12-23:
Hi,man! Thanks in advanced. When I run ./Elasticfusion with my kinect, it tells me "Your GPU "GeForce GTX 1050 Ti" isn't in the ICP Step performance database, please add it",so I run GPUTest to get GPUConfig,but it tells me "terminate called after throwing an instance of 'std::runtime_error'
  what():  Rebuild Pangolin for PNG support. I reinstall Pangolin,but it not work.Can you help me?
"

elliestath, 2018-01-18:
Hey, I have the same problem, any idea on that?

andjsmile, 2018-01-19:
@elliestath  sorry,I have no idea to solve it, and I help someone tell me how to do it.

GiorgioBullo, 2018-02-25:
hey, i found the solution. check this link:
https://github.com/stevenlovegrove/Pangolin/issues/290

closed 2022-05-27 by mp3guy