mp3guy/ElasticFusion #125: klg files

HTLife, 2017-12-23:
Here I provide the permanent link of the klg files that I converted from ICL_NUIM and TUM.

https://drive.google.com/drive/folders/1nen4wcKzB1PKhz9MiA13rPCS0QzOMwBU?usp=sharing

@mp3guy 
Maybe you could update the README.md page to include this information.  I think it will be very useful for other researchers.  And if anyone like to generate their own klg file, (https://github.com/HTLife/png_to_klg) could did that for them.

wine3603, 2018-04-28:
@HTLife Hi, Jacky, thanks! 
May I ask do you have the klg of freiburg3/walking_xyz ? I find only freiburg1 and 2 are listed in your google shares.

HTLife, 2018-04-29:
@wine3603 If it is not listed in google drive than I did not convert it.
You could convert it by yourself using [png_to_klg](https://github.com/HTLife/png_to_klg)

wine3603, 2018-05-18:
@HTLife Thanks a lot, I just did so!

root29, 2018-05-18:
Hi, I want to ask if I want to run ICL-NUIM dataset, I must use png_to_klg ?
Which format if I directly download from ICL-NUIM can run Elasticfusion?
Thank you!

HTLife, 2018-05-18:
ElasticFusion does not read image by image because of the efficiency concern.
All offline data should be converted to klg file to load into ElasticFusion.

About the format, png_to_klg could convert the following format in ICL-NUIM:
1. TUM RGB-D Compatible PNGs 
2. TUM RGB-D Compatible PNGs with noise

ayetao, 2018-10-24:
Howerver, when load a klg file generated from the project "png_to_klg" to the ElasticFusion, I get the error "Not a JPEG file: starts with 0x16 0x20". Have you ever met this problem? thank you very much.

HTLife, 2018-10-24:
It's normally caused by wrong order of association file.
You should provide depth path first.

0.033333 ./depth/scene_00_0000_rs.png 0.033333 ./rgb/scene_00_0000_rs.png

closed 2022-05-27 by mp3guy