mp3guy/ElasticFusion #127: loop closure using ground truth poses

elliestath, 2018-01-04:
I am parsing the ground truth poses (using -p while launching ElasticFusion) and getting misaligned point clouds - there is an obvious drift. 
I noticed that while using the ground truth poses, the file .freiburg saves poses identical to the ones I have used, so I guess there is no optimisation on them, they are just used as such.

I tried setting `bootstrap= true`  to use them as initial guess rather than fixed, with no results.
Is there any possibility to avoid triggering global loop closure while using my ground truth poses to check my output in that case?

Viky397, 2021-02-09:
Hi, in what format are you passing through the ground truth poses with -p?

Viky397, 2021-02-16:
Hi! I'm also trying to pass in the groundtruth poses with the -p flag, but the camera frame upon running stays static and does not move according to what I'm passing in.Below is the format my pose.txt is in. Would you happen to know if I'm missing something? Thanks!

```
1, -0.0875502211, 0.0817016153, 0.0000000000, 0.0000000000, 0.0000000000, 0.0200000000, 1.0000000000

2, -0.0875501915, 0.0817015510, 0.0000000000, 0.0000000000, 0.0000000000, 0.0200000000, 1.0000000000
```

closed 2022-05-27 by mp3guy