mp3guy/ElasticFusion #129: No devices found

mavbot66, 2018-01-11:
I have successfully build ElasticFusion on a freshly installed Ubuntu. However when I run the executable from the terminal it says : Creating live capture... failed! DeviceOpen using default: no devices found. However I am able to run dysonlab.klg file offline. Another fact is I am able to run the NIViewer in ../ElasticFusion-master/deps/OpenNI2/Bin/x64-Release. I have also linked the OpenNI2-libfreenect driver with that of the ElasticFusion i.e. /home/cvlaptop/ElasticFusion-master/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers. please help

KoSv, 2018-01-19:
Same problem here..

KoSv, 2018-01-20:
take a look here: [http://matrivian.github.io/computer%20vision/2015/04/06/openni2-2-and-nite2-2-on-ubuntu-14-04-lts.html](http://matrivian.github.io/computer%20vision/2015/04/06/openni2-2-and-nite2-2-on-ubuntu-14-04-lts.html)

shanjichao, 2018-04-04:
@KoSv @mavbot66 hello,Same problem here..Did you solve the problem? I am looking forward to your reply,thanks

shanjichao, 2018-04-04:
@KoSv @mavbot66 hello,Same problem here..Did you solve the problem? I am looking forward for your reply,thanks

KoSv, 2018-04-04:
@shanjichao I switched to windows for that ;)

shanjichao, 2018-04-05:
thanks. I maybe try for a while

wine3603, 2018-04-26:
I got this problem on my Ubuntu 16:04, with 

> Creating live capture... Warning: USB events thread - failed to set priority. This might cause loss of data...
or
> Creating live capture... failed!
> Could not open "1d27/0601@3/22": Failed to set USB interface!

Tried OpenNi2, OpenNi plus libfreenect2` No luck, 
Its weired, the same setting work for my laptop(ubuntu 14), but doesn't work on the new destop

wine3603, 2018-04-26:
HI guys, I just solved this problem on my desktop by using mp3guys Logger2 setting.
@mp3guy  mentioned to set the rule for asus live in /etc/....   I don't no why but ./Elasticfusion work 
with asus xtion pro live 0601 after setting this ....

```
> If you get an error about not being able to connect to the device follow these instructions (credit to John McCormac)
> 
>     Find the idVendor and idProduct by running lsusb (with example output):
> 
> $ lsusb
> 
> Bus 003 Device 006: ID <idVendorHere>:<idProductHere> ASUS
> 
>     Create a new usb rules file:
> 
> $ sudo editor /etc/udev/rules.d/asus.rules
> 
>     Paste in the below line with appropriate modifications (i.e. filling in the appropriate idVendor and idProduct listed above - 1d27 and 0601 for example):
> 
> SUBSYSTEM=="usb", ATTR{idVendor}=="1d27", ATTR{idProduct}=="0601", GROUP="plugdev"
> 
>     Double check you are indeed in the plugdev group, or change the group above accordingly
> 
```

LittleLampChen, 2018-08-30:
does anyone solve this problem?
it seems to be some error in device connect. i have try other code to test, failed.

Alice-kenan, 2019-03-01:
I have successfully build ElasticFusion on Ubuntu. However when I run ./ElasticFusion, the terminal says : Creating live capture... failed! Device Open using default: no devices found. However I am able to run the ./NIViewer in ~/OpenNI2/Bin/x64-Release. I have also copy the libFreenectDriver.so to ~/OpenNI2/Bin/x64-Release/OpenNI2/Drivers. please help me, Thanks a lot.

YeRen123455, 2019-06-25:
@Alice-kenan  I meet the same error, how did you solve it

closed 2022-05-27 by mp3guy